espressif/esp-matter #1367: Storing Array of Custom Struct in a Matter Attribute (esp_matter_array)

crivi30, 2025-04-11:
I am attempting to store an array of a custom struct in a Matter attribute. The struct is defined as:

```
struct adc_offset_t
{
    uint8_t channel;
    float value;
};
```

I initialize an array of adc_offset_t:

`std::vector<adc_offset_t> array = {{1, 1.23f}, {3, 1.56f}};`

The array is serialized as follows:

```
uint16_t count = array.size();
uint16_t data_size = count * sizeof(adc_offset_t);
uint8_t *val = (uint8_t *)malloc(data_size);
memcpy(val, array.data(), data_size);

// Create the Matter endpoint and cluster
endpoint_t *endpoint_zero = endpoint::get(node::get(), ENDPOINT_ZERO_ID);
cluster_t *custom_cluster = cluster::create(endpoint_zero, CONFIG_CUSTOM_CLUSTER, CLUSTER_FLAG_SERVER);

// Create the Matter attribute with the serialized data
esp_matter_attr_val_t val_adc_offset = esp_matter_array(val, data_size, count);
attribute_t *attribute_adc_offset = attribute::create(custom_cluster, CONFIG_CUSTOM_ATTRIBUTE_ADC_OFFSET, ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE, val_adc_offset);

```

This works fine for storing the data, but when i'm read it back 

```
esp_matter_attr_val_t attr_val = esp_matter_invalid(NULL);
attribute::get_val(attribute_adc_offset, &attr_val);
```
the following values are printed:

    Data size attr_val.val.a.s is 40 (correct)

    Data count attr_val.val.a.n is 40 (should be 2 instead??)

    Total size attr_val.val.a.t is 42 (correct).

**Issue When Updating the Attribute**

Later, when attempting to update the attribute, I use the following code:

```
uint16_t count = array.size();
uint16_t data_size = count * sizeof(adc_offset_t);
uint8_t *val = (uint8_t *)malloc(data_size);
memcpy(val, array.data(), data_size);

esp_matter_attr_val_t val_adc_offset = esp_matter_array(val, data_size, count);
attribute::update(ENDPOINT_ZERO_ID, CONFIG_CUSTOM_CLUSTER, CONFIG_CUSTOM_ATTRIBUTE_ADC_OFFSET, &val_adc_offset);

```
Here, I get the following error:

`E (10381) esp_matter_attribute: Error updating Endpoint 0x0000's Cluster 0x1411FC01's Attribute 0x0000000D to matter: 0x86 (CONSTRAINT_ERROR)`

It seems that something is wrong with the memory I am storing. What is causing the error 0x86 (CONSTRAINT_ERROR) when updating the attribute?

Thank you for any insight

wqx6, 2025-04-14:
The array of structure will never be stored in Matter data model. We must use AttributeAccessInterface to manage it. Please refer to https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#custom-cluster-attributes for using the attribute of array/structure types

crivi30, 2025-04-14:
Thanks for the help!

closed 2025-04-14 by crivi30