espressif/esp-matter #1369: Is there an alternative for the usage of custom clusters with Alexa, Google Home or Apple Home app?

0keanos85, 2025-04-12:
I want to transmit the amount of rain and wind direction from my device. In my understanding, custom clusters would be the first choice for this.

However, I would also like to display them in Alexa, Google Home or the Apple Home app. After looking at the existing issues, I have understood that this only works with existing clusters. Since there is currently no suitable cluster for my use case, I wanted to ask if there is another alternative for displaying them in the apps?

olavt, 2025-04-13:
As of now, the Matter standard supports clusters for temperature, humidity, and pressure but does not have specific clusters for precipitation (rain) or wind speed. Manufacturers can work around this with custom clusters, but these measurements won’t be natively interoperable within Matter ecosystems. If you’re building or choosing a Matter-compatible weather station, verify with the manufacturer whether rain and wind data are exposed via Matter (likely as custom attributes) or require a separate app/protocol.

For non-standard data, I think you need to use a custom app to display the data properly. It's a bit strange that Matter yet does not support weather station devices properly.

0keanos85, 2025-04-21:
Yes, that is kinda what I thougt. Thanks for confirming my assumption!

I asked the question about the support of precipitation and wind speed in the connectedhomeip repo and there it was explained that Matter is for interoperable, certifiable Smart Home products and participants have not brought forward those features yet.

[https://github.com/project-chip/connectedhomeip/issues/38375](url)

I will close this Issue as my question has been answered.

closed 2025-04-21 by 0keanos85