espressif/esp-matter #1371: Enable user BLE service and broadcasting

MaplestoryAlen, 2025-04-15:
IDF：V5.2.3
MATTER: V1.4
 
    I want to manually turn off the distribution network when opening the distribution network window in Matter, and then turn on my own BLE service. If I don't initialize Matter's Bluetooth, my BLE service can still broadcast normally. However, if I first initialize from Matter and then turn off Matter's Bluetooth, the broadcast cannot be turned on after initializing my own Bluetooth. I don't know where I did it wrong
   Because I am in IDF version 5.2.3, the "ble_gatts_free_svcs()" function was not found, so I used "ble_gatts_stop()" and "ble_gatts_reset()" in the same way.

void StartUserBleDegbugMode(void)
{   
  
      MatterCloseCommissioningWindow(); 
     ClearCHIPoGattService();
      IotBleStart();
  
}

void ClearCHIPoGattService(void)
{
    chip::DeviceLayer::Internal::BLEMgrImpl().StopCHIPoBleGattService();
}
void BLEManagerImpl::StopCHIPoBleGattService(void)
{
     CancelBleAdvTimeoutTimer();	
     ClearScanResponseData();
     if (ble_gap_adv_active())
     {
	  ble_gap_adv_stop();
	 vTaskDelay(pdMS_TO_TICKS(300));
     }
     ble_gatts_stop(); 
     DeinitBLE();	 
}

log:

[image]

[image]

MaplestoryAlen, 2025-05-09:
Is there an official technician to answer!

rahult-github, 2025-05-15:
Hello @MaplestoryAlen  ,

May i know the exact application changes you are trying , so that we can try the same at our end to reproduce ? 

[image]

The steps mentioned here are not very clear.  e.g. attempting to add services before stack is running is not possible , but in this you have mentioned that you add services without any mention of nimble_port_init .  So , to avoid any miscommunication, request to please share code or a sample mini project that we can use to reproduce and check more. 

I tried a sample application where i did stack init / deinit in a loop and tried to add services in between of this. I do see it working fine. So i feel there may be issue in the sequence of steps or some missing step .  Sample code for your reference below which works ok. 

static void stack_init_deinit(void)
{
    int rc;
    while(1) {

        vTaskDelay(400);

        ESP_LOGI(tag, "Deinit host");

        rc = nimble_port_stop();
        if (rc == 0) {
            nimble_port_deinit();
        } else {
            ESP_LOGI(tag, "Nimble port stop failed, rc = %d", rc);
            break;
        }

       vTaskDelay(400);

        ESP_LOGI(tag, "Init host");

        rc = nimble_port_init();
       if (rc != ESP_OK) {
            ESP_LOGI(tag, "Failed to init nimble %d ", rc);
            break;
        }

       ble_gatts_stop();

       ble_gatts_reset();

       gatt_svr_init();

        nimble_port_freertos_init(bleprph_host_task);

        ESP_LOGI(tag, "Waiting for 1 second");
   }
}

MaplestoryAlen, 2025-05-15:
> nimble_port_freertos_init

    Thank you very much for your answer. I have tested the method you provided and found that after BLE broadcasting is enabled, there will be an error message that cannot be released. If you want to turn off the matter service and then restart the user's own Bluetooth service after matter BLE initialization, there will be a conflict. You can only select matter service during initialization or user-defined service

pimpalemahesh, 2025-05-16:
@MaplestoryAlen Just wanted to confirm—have you disabled the CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING option?

MaplestoryAlen, 2025-05-20:
> [@MaplestoryAlen](https://github.com/MaplestoryAlen) Just wanted to confirm—have you disabled the CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING option?
  CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n

pimpalemahesh, 2025-05-23:
@MaplestoryAlen Could you try this patch [custom_ble_service_over_matter_release_v1_4_idf_v5_2_3.txt](https://github.com/user-attachments/files/20408693/custom_ble_service_over_matter_release_v1_4_idf_v5_2_3.txt)

Please set the following options using menuconfig:
```
CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n
CONFIG_BT_NIMBLE_DYNAMIC_SERVICE=y
```
This patch includes resetting the GATT service after Matter commissioning, adding a custom service dynamically using the `ble_gatts_add_dynamic_svcs() `API, and starting advertisement

MaplestoryAlen, 2025-06-17:
> [@MaplestoryAlen](https://github.com/MaplestoryAlen) Could you please clarify your exact requirements? The above patch should work if your goal is to run your own BLE service on the same stack after commissioning is complete.
> 
> Or, are you looking to start your custom BLE service right from boot-up?

Because there are many user-defined Bluetooth GATT services, I don't want to add user-defined services to Matter's GATT service. Therefore, I want Matter's Bluetooth and user-defined Bluetooth to work independently, turning off user-defined Bluetooth services when Matter Bluetooth is running, or turning off Bluetooth services on Matter's side when user-defined Bluetooth services need to work