espressif/esp-matter #1372: How to populate the Rloc16 attribute of the Thread Networking Diagnostics cluster?

olavt, 2025-04-15:
How can I properly populate this attribute?

[image]

wqx6, 2025-04-23:
This attribute is provisional and was added in https://github.com/project-chip/connectedhomeip/commit/0435166c647e9d3b25bab9c7fd4cf25c0bc1dce3, but the connectedhomeip repo in esp_matter has not been updated to the last master branch so it doesn't include that commit.

wqx6, 2025-05-15:
You can get it by calling `otThreadGetRloc16(esp_openthread_get_instance())`

wqx6, 2025-08-20:
The RLOC16 attribute is not managed by our data model, it is managed by the AttributeAccessInterface(See https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/thread-network-diagnostics-server/thread-network-diagnostics-server.cpp#L134), so you cannot use update the attribute value using `attribute::updating()`.

olavt, 2025-08-22:
@wqx6 Why are attributes managed differently. It's quite confusing to understand the differences and when it's one way or another.