espressif/esp-matter #1373: troubles with gaving a name to endpoints

ravencarcass, 2025-04-15:
Hello, 
I'm making a Matter project where I used the example code "light". 
I added extra endpoints, but it's ugly in the google home app.
I would like to change the name for the app.
I found here on GitHub an example and tried it on my project but it didn't work out.
Can some body help me? 

This is the code for one endpoint. 
```
esp_err_t err = ESP_OK;
nvs_flash_init();

app_driver_handle_t light_handle = app_driver_light_init();
periferal_init();
node::config_t node_config;

node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
ABORT_APP_ON_FAILURE(node != nullptr, ESP_LOGE(TAG, "Failed to create Matter node"));

dimmable_light::config_t light_config;
light_config.on_off.on_off = DEFAULT_POWER;
light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;

constexpr uint8_t kNamespaceHorizontalControl = 60;
constexpr uint8_t KTagHorizontal = 0;

const Descriptor::Structs::SemanticTagStruct::Type horizontalControlTagList[] = {
    {.namespaceID = kNamespaceHorizontalControl, .tag = KTagHorizontal}};

endpoint_t *endpoint = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create endpoint"));

SetTagList(endpoint::get_id(endpoint), chip::Span<const Descriptor::Structs::SemanticTagStruct::Type>(horizontalControlTagList));
light_endpoint_id = endpoint::get_id(endpoint);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);
```

olavt, 2025-04-18:
Endpoints in Matter does not have names.

"I added extra endpoints, but it's ugly in the google home app."

What exactly looks ugly.

ravencarcass, 2025-04-18:
They are all named like test_product(x). I would like to bring s bit more logic into it.

Jerry-ESP, 2025-04-21:
@ravencarcass Before call SetTagList(), you need to add the taglist feature: https://github.com/espressif/esp-matter/blob/211d2761e528d2edeb15a5c215cd2ad5c37e99e3/examples/generic_switch/main/app_main.cpp#L139

dhrishi, 2025-05-14:
@ravencarcass Please close the issue if resolved