espressif/esp-matter #1374: Cannot update or display CumulativeEnergyImported due to array type mismatch

OHyungM, 2025-04-16:
I’m developing a Matter-based energy monitoring device (plug) using the ESP-Matter SDK.
In my application, I have two values:

1. Power consumption (instantaneous): int64, working well
2. Total cumulative energy: originally defined as an array in the ElectricalEnergyMeasurement cluster

I extract energy data from my own energy metering code and try to push it to the Matter stack using:
-> attribute::update(endpoint_id, 0x0091, 0x0001, &energy_val);
Where energy_val is of type ESP_MATTER_VAL_TYPE_ARRAY.

However, I am seeing the following log when trying to update the cumulative energy:
I (20560359) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000091's Attribute 0x00000001 is <invalid type: 4> **********
E (20560369) esp_matter_attribute: Insufficient space for reading Endpoint 0x0001's Cluster 0x00000091's Attribute 0x00000001: required: 4, max: 2

At the same time, this attribute is not shown in the SmartThings app, even though the power value appears correctly.

Questions:
1. What is the correct structure and format for sending an array-based attribute like CumulativeEnergyImported via attribute::update()?
   1-1 . How should the .val.a.b, .s, .n, .t fields be set up to match the cluster definition?

2. If the array structure is required, is there any supported way to display this value in the SmartThings app UI?

* My current code snippet for update::

uint8_t energy_arr[4];
uint32_t energy_val_raw = powermeter.power_value.energy;
uint32_t energy_be = htonl(energy_val_raw);
memcpy(energy_arr, &energy_be, sizeof(energy_arr));

esp_matter_attr_val_t energy_val = {
    .type = ESP_MATTER_VAL_TYPE_ARRAY,
    .val.a = {
        .b = energy_arr,
        .s = sizeof(energy_arr),
        .n = 1,
        .t = sizeof(energy_arr)
    }
};

esp_err_t err = attribute::update(endpoint_id, 0x91, 0x01, &energy_val);

Thank you in advance for your help!

olavt, 2025-04-16:
You are encountering this issue because the CumulativeEnergyImported attribute (ID 0x0001) in the Matter ElectricalEnergyMeasurement cluster is defined as an array of structures (specifically, an array of EnergyMeasurementStruct), not a simple byte array or a single integer. The ESP-Matter SDK expects the attribute value to match the Matter spec's structure.

OHyungM, 2025-04-16:
> You are encountering this issue because the CumulativeEnergyImported attribute (ID 0x0001) in the Matter ElectricalEnergyMeasurement cluster is defined as an array of structures (specifically, an [...]

Thank you for the clarification!

Based on your response, I understand now that the CumulativeEnergyImported attribute (ID 0x0001) is defined as an array of EnergyMeasurementStruct, not a simple byte array or integer. That makes sense.

However, I’d like to ask for guidance on how to structure this data correctly in ESP-Matter.

1. What is the exact format of EnergyMeasurementStruct expected by the ESP-Matter SDK?
Could you provide the definition of the struct, including field types and sizes?

2. How should I construct a valid esp_matter_attr_val_t to represent an array of EnergyMeasurementStruct?
For example:

Should .val.a.b point to a packed array of structs?

What should .s, .n, and .t be set to for a single struct entry?

3. If I only want to report a single energy measurement (e.g. totalWh), is it valid to send an array with n = 1 and a single struct entry?

A small code snippet or documentation reference showing the correct way to construct and send an array of EnergyMeasurementStruct in ESP-Matter using attribute::update().

Thanks again for your support!

olavt, 2025-04-17:
Take a look at the Matter specification document: "Matter Application Cluster Specification Version 1.4" under section "2.12.5.1. EnergyMeasurementStruct Type".

closed 2025-05-07 by OHyungM