espressif/esp-matter #1379: When the device sets the ColorTemperatureMireds specific value, the color of the UI in Apple Home does not change

wolen666, 2025-04-19:
1. Below is my main code. When ColorTemperatureMireds is greater than 500, the UI color is warm white, and below 100 it is cold white. I first clicked on cold white in the Apple Home UI, and when the test task was scheduled to set ColorTemperatureMireds to warm white, I found that the Apple Home UI always displayed cold white. When I exited the home and reopened the UI, it displayed normally, which was very strange.
2. matter1-3
3. esp-idf5.1

void MatterWork_UpdateColorTemp(intptr_t context)
{
    auto * work   = reinterpret_cast<ColorTempWork *>(context);
    auto   ep     = work->mEndpoint;
    auto   mireds = work->mMireds;
    delete work;

    {
        chip::DeviceLayer::StackLock lock;
        chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Set(ep, mireds);
    }

    auto * imEngine        = chip::app::InteractionModelEngine::GetInstance();
    auto & reportingEngine = imEngine->GetReportingEngine();
    chip::app::AttributePathParams path(ep,
                                        chip::app::Clusters::ColorControl::Id,
                                        chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Id);
    reportingEngine.SetDirty(path);  // :contentReference[oaicite:7]{index=7}

    reportingEngine.ScheduleRun();   // :contentReference[oaicite:8]{index=8}
}

static void AppTest(void* pvParameters)
{
    bool     update_val         = true;
    bool     temperature_update = true;
    uint16_t temper_up          = 500;
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(5000));
  

        if (temperature_update) {
            uint16_t current = 0;
            {
                chip::DeviceLayer::StackLock lock;
                chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Get(1, &current);
            }
            printf("currentColorTemperatureMireds : %u mireds\n", current);
            vTaskDelay(pdMS_TO_TICKS(8000));
        }

        if (update_val) {
            printf("updateColorTemperatureMireds: %u mireds\n", temper_up);

            auto * work = new ColorTempWork{ .mEndpoint = 1, .mMireds = temper_up };

            chip::DeviceLayer::PlatformMgr().ScheduleWork(
                MatterWork_UpdateColorTemp,
                reinterpret_cast<intptr_t>(work)
            );  // :contentReference[oaicite:9]{index=9}

            temper_up += 10;
            vTaskDelay(pdMS_TO_TICKS(4000));
        }
    }
}

jadhavrohit924, 2025-04-21:
@wolen666 Please attach the complete logs in a file where you see an issue. Please share what are you using esp-matter or connectedhomeip, to reproduce and debug the issue.

wolen666, 2025-04-22:
premise
   esp-idf 5.1 
   connectedhomeip---->commit 14b332966cde23ea425e99ebfe5dcd7ef7dd076d (HEAD -> v1.3-branch, origin/v1.3-branch)
 
1.Below is my code, which changes the color temperature value at regular intervals.Modified the color temperature upload value of the previous code
static void AppTest(void* pvParameters)
{
    bool     temperature_update = false;
    uint16_t temper_up          = 250;

    while (1)
    {
        if (network) 
        {
            vTaskDelay(pdMS_TO_TICKS(4000));
            network             = false;
            temperature_update = true;
        }

        if (temperature_update) 
        {
            uint16_t current = 0;
            chip::DeviceLayer::StackLock lock;  
            chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Get(1, &current);

            temper_up += 10;
            chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Set(1, temper_up);

            auto & reportingEngine = InteractionModelEngine::GetInstance()->GetReportingEngine();
            AttributePathParams path(1,
                                    chip::app::Clusters::ColorControl::Id,
                                    chip::app::Clusters::ColorControl::Attributes::ColorTemperatureMireds::Id);
            reportingEngine.SetDirty(path);
            reportingEngine.ScheduleRun();  
        
            chip::DeviceLayer::StackUnlock unlock;

            printf("current: %u mireds\n", current);
            printf("[update] → ColorTemperatureMireds: %u mireds\n", temper_up);
        }

        vTaskDelay(pdMS_TO_TICKS(6000));
    }
}

2.Setting zap, according to the matter standard document description kelvins = 1,000,000 / ColorTemperatureMireds, so ColorTempPhysicalMinMireds is 370, ColorTempPhysicalMinMireds is 135.

[image]

3. Add the device to Apple Home, and the log displayed is as follows

I (335345) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x300', EndPoint ID: '0x1', Attribute ID: '0x7'
I (335355) light-app-callbacks: Current free heap: 114340

current: 280 mireds
[update] → ColorTemperatureMireds: 290 mireds
I (335375) chip[EM]: <<< [E:21655i S:65168 M:124660557] (S) Msg TX to 1:000000007FD6986A [B852] [UDP:192.168.5.125:57522] --- Type 0001:05 (IM:ReportData)
I (335405) chip[EM]: >>> [E:21655i S:65168 M:227711312 (Ack:124660557)] (S) Msg RX from 1:000000007FD6986A [B852] --- Type 0001:01 (IM:StatusResponse)
I (335415) chip[IM]: Received status response, status is 0x00
I (335415) chip[EM]: <<< [E:21655i S:65168 M:124660558 (Ack:227711312)] (S) Msg TX to 1:000000007FD6986A [B852] [UDP:192.168.5.125:57522] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (341375) all-clusters-app: Attribute changed on non-Matter task 'app_test'

I (341375) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x300', EndPoint ID: '0x1', Attribute ID: '0x7'
I (341385) light-app-callbacks: Current free heap: 114340

current: 290 mireds
[update] → ColorTemperatureMireds: 300 mireds
I (341405) chip[EM]: <<< [E:21656i S:65168 M:124660559] (S) Msg TX to 1:000000007FD6986A [B852] [UDP:192.168.5.125:57522] --- Type 0001:05 (IM:ReportData)
I (341595) chip[EM]: >>> [E:21656i S:65168 M:227711313 (Ack:124660559)] (S) Msg RX from 1:000000007FD6986A [B852] --- Type 0001:01 (IM:StatusResponse)
I (341605) chip[IM]: Received status response, status is 0x00
I (341615) chip[EM]: <<< [E:21656i S:65168 M:124660560 (Ack:227711313)] (S) Msg TX to 1:000000007FD6986A [B852] [UDP:192.168.5.125:57522] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (343235) chip[EM]: >>> [E:19589r S:65168 M:227711314] (S) Msg RX from 1:000000007FD6986A [B852] --- Type 0001:08 (IM:InvokeCommandRequest)
I (343245) chip[ZCL]: OpCreds: Received an UpdateFabricLabel command
I (343255) chip[FP]: Metadata for Fabric 0x1 persisted to storage.
I (343255) chip[EM]: <<< [E:19589r S:65168 M:124660561 (Ack:227711314)] (S) Msg TX to 1:000000007FD6986A [B852] [UDP:192.168.5.125:57522] --- Type 0001:09 (IM:InvokeCommandResponse)

4.When it is set to 162, the UI value is cold. After a while, the apptest task will upload a value of 300. At this time, the UI color should change to warm color. In fact, the UI color does not change. If I exit the Apple homepage and re-enter the UI color, the UI color will change.I think this color change should respond immediately

https://github.com/user-attachments/assets/40dde88a-3504-44da-acab-c25f603c6ec1

jadhavrohit924, 2025-04-24:
From your logs, I see that the attribute `Set` is working fine, only the UI of Apple Home App is not updated. This looks like an issue could be with the Apple Home App.

closed 2025-04-30 by wolen666