espressif/esp-matter #1381: How to report Accumulated Energy data to SmartThings using the ElectricalEnergyMeasurement cluster?

OHyungM, 2025-04-21:
Hi,
I’m working on a Matter-based energy monitoring plug device using the ESP-Matter SDK.
I want to display total cumulative energy consumption (in Wh or mWh) in the SmartThings app UI.

I’m trying to understand:

1. Should I use the CumulativeEnergyImported attribute (ID: 0x0001) of the ElectricalEnergyMeasurement cluster (ID: 0x0091) for this?

Is this the standard attribute intended for reporting total energy used by a device?

2. How should I update this attribute correctly using ESP-Matter?

Is there an example of how to build and send the EnergyMeasurementStruct required for this attribute?

The attribute expects an array or struct format — is this supported by esp_matter_attr_val_t?

3. Is there an official Matter specification document (PDF or HTML) I can reference for this cluster and attribute?

Specifically: format of EnergyMeasurementStruct and update behavior for this attribute

To send cumulative energy usage to SmartThings in a way that will visibly appear in the app UI using a Matter-compliant approach.

Any guidance or sample implementation would be greatly appreciated. Thank you!

jadhavrohit924, 2025-04-21:
Hi @OHyungM, 
> 1 Should I use the CumulativeEnergyImported attribute (ID: 0x0001) of the ElectricalEnergyMeasurement cluster (ID: 0x0091) for this?
Is this the standard attribute intended for reporting total energy used by a device?

Yes, you can use the attribute to get total energy consumption.

> 2 How should I update this attribute correctly using ESP-Matter?
Is there an example of how to build and send the EnergyMeasurementStruct required for this attribute?

Please refer the reference implementation of ElecricalEnergyMeasurement cluster in [connectedhomeip](https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/src/electrical-energy-measurement-stub.cpp) and in [esp-matter](https://github.com/espressif/esp-matter/issues/1102#issuecomment-2408704757).

> 3 Is there an official Matter specification document (PDF or HTML) I can reference for this cluster and attribute?
Specifically: format of EnergyMeasurementStruct and update behavior for this attribute

Please download the latest [Matter Specification](https://csa-iot.org/developer-resource/specifications-download-request/) to know about EnergyMeasurementStruct.

> To send cumulative energy usage to SmartThings in a way that will visibly appear in the app UI using a Matter-compliant approach.

We are not sure if SmartThings supports the EnergyMeasurementCluster. We will check and let you know.

OHyungM, 2025-04-21:
Thank you for your reply @jadhavrohit924,

Let me ask you one more question.

Is it correct to use that attribute even if I am updating my measured data, not just fetching the total accumulated power?

I am getting the values ​​through a sensor that measures power separately.

jadhavrohit924, 2025-04-22:
I thinks that's okay.

closed 2025-05-07 by OHyungM