espressif/esp-matter #1382: i'm new using matter i wanna create flow sensor matter composent that indicates the flow measurement dynamically so i used light example so i find problems with functions call back and update

gabask, 2025-04-21:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

**Additional context**
...

jadhavrohit924, 2025-05-06:
@gabask Please add some more details of the issue, your working environment, did you change anything in the default light example? If yes, please share the patch. What is your actual issue? Please add reproduction steps.

gabask, 2025-05-06:
@jadhavrohit924 hi i think i fix the problem i will share the code with y and correct it if there's  mistake plz especially in app_driver.cpp i try to simulate flow measurement and return it in home assistant interface  as first step
**********************this is app_main.cpp*****************************
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_attribute.h>
#include <esp_matter_cluster.h>
#include <esp_matter_endpoint.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>

#include <common_macros.h>
#include <app_priv.h>
#include <app_reset.h>
#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
#include <platform/ESP32/OpenthreadLauncher.h>
#endif

#include <app/server/CommissioningWindowManager.h>
#include <app/server/Server.h>

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
#include <esp_matter_providers.h>
#include <lib/support/Span.h>
#ifdef CONFIG_SEC_CERT_DAC_PROVIDER
#include <platform/ESP32/ESP32SecureCertDACProvider.h>
#elif defined(CONFIG_FACTORY_PARTITION_DAC_PROVIDER)
#include <platform/ESP32/ESP32FactoryDataProvider.h>
#endif
using namespace chip::DeviceLayer;
#endif

static const char *TAG = "app_main";
uint16_t flow_endpoint_id = 1;

using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace chip::app::Clusters;

constexpr auto k_timeout_seconds = 300;

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
extern const uint8_t cd_start[] asm("_binary_certification_declaration_der_start");
extern const uint8_t cd_end[] asm("_binary_certification_declaration_der_end");

const chip::ByteSpan cdSpan(cd_start, static_cast<size_t>(cd_end - cd_start));
#endif // CONFIG_ENABLE_SET_CERT_DECLARATION_API

#if CONFIG_ENABLE_ENCRYPTED_OTA
extern const char decryption_key_start[] asm("_binary_esp_image_encryption_key_pem_start");
extern const char decryption_key_end[] asm("_binary_esp_image_encryption_key_pem_end");

static const char *s_decryption_key = decryption_key_start;
static const uint16_t s_decryption_key_len = decryption_key_end - decryption_key_start;
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
[... 214 lines cut ...]
        } 
    }
  return err;
}*/

esp_err_t app_driver_flow_set_defaults(uint16_t endpoint_id)
{
    esp_err_t err = ESP_OK;

    // Simulated flow rate in deci-liters per minute (e.g., 73 = 7.3 L/min)
    int16_t simulated_flow_value = 100;
    esp_matter_attr_val_t val = esp_matter_int16(simulated_flow_value);

    // Get the MeasuredValue attribute
    attribute_t *attribute = attribute::get(endpoint_id, FlowMeasurement::Id, FlowMeasurement::Attributes::MeasuredValue::Id);
    if (attribute != nullptr) {
        err |= attribute::set_val(attribute, &val);
        ESP_LOGI(TAG, "Flow sensor simulated value set to: %d", simulated_flow_value);
    } else {
        ESP_LOGE(TAG, "Failed to get MeasuredValue attribute");
        err |= ESP_FAIL;
    }

    return err;
}

closed 2025-05-12 by gabask