espressif/esp-matter #1383: DoorStateChange event support

sdrasti, 2025-04-22:
When creating test plans with the PICSgenerator tool, it is stated in the Events-Section that the DoorStateChange Event is not supported:

```xml
<!--Events PICS write-->
        <events>
            <picsItem>
                <itemNumber>DRLK.S.E00</itemNumber>
                <feature>Does the device implement DoorLockAlarm event?</feature>
                <reference>130.2.4. Events - allclusters.html[pdf]</reference>
                <status cond="DRLK.S">M</status>
                <support>true</support>
            </picsItem>
            <picsItem>
                <itemNumber>DRLK.S.E01</itemNumber>
                <feature>Does the device implement DoorStateChange event?</feature>
                <reference>130.2.4. Events - allclusters.html[pdf]</reference>
                <status cond="DRLK.S AND DRLK.S.F05">M</status>
                <support>false</support>
            </picsItem>
            <picsItem>
                <itemNumber>DRLK.S.E02</itemNumber>
                <feature>Does the device implement LockOperation event?</feature>
                <reference>130.2.4. Events - allclusters.html[pdf]</reference>
                <status cond="DRLK.S">M</status>
                <support>true</support>
            </picsItem>
            <picsItem>
                <itemNumber>DRLK.S.E03</itemNumber>
                <feature>Does the device implement LockOperationError event?</feature>
                <reference>130.2.4. Events - allclusters.html[pdf]</reference>
                <status cond="DRLK.S">M</status>
                <support>true</support>
            </picsItem>
            <picsItem>
                <itemNumber>DRLK.S.E04</itemNumber>
                <feature>Does the device implement LockUserChange event?</feature>
                <reference>130.2.4. Events - allclusters.html[pdf]</reference>
                <status cond="DRLK.S AND DRLK.S.F08">M</status>
                <support>false</support>
            </picsItem>
        </events>
```

This is how the DPS feature is added in the code, and with that I can successfully read and write the door state attribute:

```c
 /* Add optional DPS feature */
      cluster_t *DoorLockCluster = cluster::get(Endpoint1, DoorLock::Id);
      Result                     = cluster::door_lock::feature::door_position_sensor::add(DoorLockCluster);
```

From the eep-matter components side It seems as if event is created when adding the feature:
https://github.com/espressif/esp-matter/blob/211d2761e528d2edeb15a5c215cd2ad5c37e99e3/components/esp_matter/esp_matter_feature.cpp#L3874

Is there anything else I should consider during the DPS feature instantiation?

Jerry-ESP, 2025-04-22:
You can use this chip-tool command to read the event: ./chip-tool doorlock read-event door-state-change 1 0 .

The esp-matter sdk will add the door_state_change event when add the door_position_sensor feature, the reason why the event not enable in XML file should be the event list attribute was removed, the PICSgenerator can't get the actually supported events in the doorlock, you can enable the event manually in the XML file according to your device's implementation.

sdrasti, 2025-04-22:
@Jerry-ESP thank you! Reading the event with the chip-tool also works as expected, so it's done with editing the XML accordingly.

closed 2025-04-22 by sdrasti