espressif/esp-matter #1386: ICD App Goes to sleep to quickly and does not allow an interview to be completed

laptopfreek0, 2025-04-22:
**Describe the bug**
When using the ICD app with home assistant during the final step of "Connecting device to Home Assistant" the esp32-c6 goes to sleep too quickly and does not allow for the interview to complete. Every few attempts after a factory reset, the esp stays awake long enough to complete the commission, sometimes with a failure in HomeAssistant, sometimes with success. The same does not happen while commissioning a non ICD app.

**Environment**
- ESP-Matter Commit Id: 71af1724f07725f5e58a93e2ec3317bebf2d38f8 (esp-matter v1.4 release)
- ESP-IDF Commit Id: c9763f62dd00c887a1a8fafe388db868a7e44069 (esp idf v5.2.3 release)
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):
```log
I (23) boot: ESP-IDF v5.2.3 2nd stage bootloader
I (23) boot: compile time Apr 21 2025 10:51:41
I (24) boot: chip revision: v0.1
I (26) boot: efuse block revision: v0.3
I (31) boot.esp32c6: SPI Speed      : 80MHz
I (36) boot.esp32c6: SPI Mode       : DIO
I (40) boot.esp32c6: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (69) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (76) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (84) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (91) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (99) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (106) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (114) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (121) boot: End of partition table
I (126) esp_image: segment 0: paddr=00020020 vaddr=42120020 size=3bf38h (245560) map
I (181) esp_image: segment 1: paddr=0005bf60 vaddr=40800000 size=040b8h ( 16568) load
I (185) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=113cc4h (1129668) map
I (402) esp_image: segment 3: paddr=00173cec vaddr=408040b8 size=1661ch ( 91676) load
I (422) esp_image: segment 4: paddr=0018a310 vaddr=4081a6e0 size=02cb8h ( 11448) load
I (426) esp_image: segment 5: paddr=0018cfd0 vaddr=50000000 size=00048h (    72) load
I (434) boot: Loaded app from partition at offset 0x20000
I (435) boot: Disabling RNG early entropy source...
I (451) cpu_start: Unicore app
W (460) clk: esp_perip_clk_init() has not been implemented yet
I (467) cpu_start: Pro cpu start user code
I (467) cpu_start: cpu freq: 160000000 Hz
I (467) cpu_start: Application information:
I (470) cpu_start: Project name:     icd_app
I (475) cpu_start: App version:      1.0
I (479) cpu_start: Compile time:     Apr 21 2025 16:31:21
I (485) cpu_start: ELF file SHA256:  47e7962c5...
I (491) cpu_start: ESP-IDF:          v5.2.3
I (496) cpu_start: Min chip rev:     v0.0
I (501) cpu_start: Max chip rev:     v0.99 
I (506) cpu_start: Chip rev:         v0.1
I (510) heap_init: Initializing. RAM available for dynamic allocation:
I (518) heap_init: At 40832AC0 len 00049B50 (294 KiB): RAM
I (524) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (530) heap_init: At 50000048 len 00003FA0 (15 KiB): RTCRAM
I (537) spi_flash: detected chip: generic
I (541) spi_flash: flash io: dio
I (546) sleep: Enable automatic switching of GPIO sleep configuration
I (552) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (571) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (572) coexist: coex firmware version: 27d8387
I (575) coexist: coexist rom version 5b8dcfa
I (581) main_task: Started on CPU0
I (584) main_task: Calling app_main()
I (597) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 160, APB_MIN: 160, Light sleep: ENABLED
I (601) Driver: Creating Button
I (602) Driver: Creating Button Not Null
I (607) button: IoT Button Version: 3.5.0
I (612) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (622) app_main: Generic Switch created with endpoint_id 1
I (627) Driver: Creating Button
I (631) Driver: Creating Button Not Null
I (635) button: IoT Button Version: 3.5.0
I (640) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (650) Driver: Creating Button
I (653) Driver: Creating Button Not Null
I (658) button: IoT Button Version: 3.5.0
I (662) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (672) app_main: Generic Switch created with endpoint_id 2
I (678) Driver: Creating Button
I (682) Driver: Creating Button Not Null
I (686) button: IoT Button Version: 3.5.0
I (691) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (700) Driver: Creating Button
I (704) Driver: Creating Button Not Null
I (709) button: IoT Button Version: 3.5.0
I (713) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (723) app_main: Generic Switch created with endpoint_id 3
I (729) Driver: Creating Button
I (733) Driver: Creating Button Not Null
I (737) button: IoT Button Version: 3.5.0
I (742) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (751) Driver: Creating Button
I (755) Driver: Creating Button Not Null
I (759) button: IoT Button Version: 3.5.0
I (764) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (774) app_main: Generic Switch created with endpoint_id 4
I (780) Driver: Creating Button
I (783) Driver: Creating Button Not Null
I (788) button: IoT Button Version: 3.5.0
I (793) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (803) battery: battery sensor initialized successfully
I (808) button: IoT Button Version: 3.5.0
I (813) gpio: GPIO[7]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (833) chip[DL]: NVS set: chip-counters/reboot-count = 2 (0x2)
I (834) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
I (837) BLE_INIT: Using main XTAL as clock source
I (838) esp openthread sleep: Enable ieee802154 light sleep, the wake up source is ESP timer
I (851) OPENTHREAD: Host connection mode none
I (856) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11
I (850) BLE_INIT: ble controller commit:[7491a85]
W (868) BLE_INIT: BLE modem sleep is enabled
I (874) BLE_INIT: Modem BLE MAC retention initialization
W (880) BLE_INIT: Enable light sleep, the wake up source is BLE timer
I (886) BLE_INIT: Bluetooth MAC: 8c:bf:ea:b4:40:26
W (924) phy_init: saving new calibration data because of checksum failure, mode(0)
I (936) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
I (936) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (944) NimBLE: GAP procedure initiated: stop advertising.

I (946) NimBLE: Failed to restore IRKs from store; status=8

I (952) CHIP[DL]: BLE host-controller synced
I (967) OPENTHREAD: OpenThread attached to netif
I (969) chip[DL]: OpenThread started: OK
I (971) chip[DL]: Setting OpenThread device type to SLEEPY END DEVICE
I(974) OPENTHREAD:[N] Mle-----------: Mode 0x0c -> 0x04 [rx-on:no ftd:no full-net:no]
None
Waiting for the device to reconnect...
```
- Host Machine OS: Ubuntu 24.04
- Host Machine Python version: 3.10.12

- Commissioner app and versions if present: HomeAssistant Latest
- Home hub app and versions if present: 
- Commissioner's logs if present:

**Any additional details**
...

Jerry-ESP, 2025-04-23:
Please post the whole logs for the commissioning failed, start from device power up.

laptopfreek0, 2025-04-23:
I have attached as complete as logs as I can. It seems the ICD app doesn't log as much during the commissioning window. I have also attached the homeassistant core matter server logs as well. Let me know if there are any other logs or things you would like me to try.

[core_matter_server_relavent.log](https://github.com/user-attachments/files/19869426/core_matter_server_relavent.log)

[log.icd_app.20250423091340.txt](https://github.com/user-attachments/files/19869485/log.icd_app.20250423091340.txt)

Jerry-ESP, 2025-04-24:
Sorry, I don't understand the device's work flow, I haven't found any commissioning process from the log, even ble connection established. Can you describe your device more detail, or can you test the device using chip-tool?  And can you please share the sdkconfig file?

laptopfreek0, 2025-04-25:
My observations are when I enable the ICD server the logging is vastly cut down and I cannot find a setting to turn them back up so I can see what's going on. But I do see that during the commissioning step the ESP goes to sleep and turns off the UART monitoring, while the commissioning server is still attempting to "add device to home assistant".

I have uploaded my code including the sdkconfig to [github](https://github.com/laptopfreek0/esp-matter-icd-with-factor-reset)

To date, I haven't taken time to learn the chip-tool, but will do so to attempt to test the device commissioning that way. 

Another thing of note. During commissioning the ESP does end up on the thread network, and it appears to properly work and communicate with the other nodes, but does not get added into the homeassistant matter server.

Jerry-ESP, 2025-04-27:
I think you can test the [icd-app](https://github.com/espressif/esp-matter/tree/main/examples/icd_app) example in esp-matter first.

laptopfreek0, 2025-04-27:
The version I uploaded to github is the same as the icd-app except I added a reset button so I didn't have to continue to erase_flash and re-flash every time commissioning failed. The same issue occurs with the icd-app. Commissioning gets as far as getting on the tread network, and then responding to the matter server when pinged, but during the interview step the esp goes to sleep without completing the Adding device to homeassistant step. In both instances if I turn off the ICD server commissioning succeeds every time.

Jerry-ESP, 2025-04-30:
[icd_app_merge.zip](https://github.com/user-attachments/files/19974560/icd_app_merge.zip)
I tested the icd_app on my side, the log can output when commissioning, please try the image in the attachment, just flash it at 0x0 address.

laptopfreek0, 2025-05-01:
So I attempted to flash and I was met with the following error `Unexpected chip id in image. Expected 13 but value was 16. Is this image for a different chip model?
`
I flashed it anyway and it complains about `Invalid chip id. Expected 13 read 16. Bootloader for wrong chip?`

Prior to flash I noticed that my ESP32 is the following model: "Chip is ESP32-C6FH4 (QFN32) (revision v0.1)"

Jerry-ESP, 2025-05-06:
[icd_app_c6_merge.zip](https://github.com/user-attachments/files/20052023/icd_app_c6_merge.zip)
Please use this firmware for testing, the one before is for esp32h2, and the current one is for esp32c6 chip.

laptopfreek0, 2025-05-15:
@Jerry-ESP  I have flashed the bin file that you provided and every time I commission it works perfectly. I noticed that the bin file was created using esp-idf v5.3.2, and I compiled with the recommended v5.2.3 Could that cause the discrepancy? Could you share the skdconfig, and specifications to see if I can upgrade my project to work properly like the bin file you provided?

Jerry-ESP, 2025-05-16:
[icd_config_bin.zip](https://github.com/user-attachments/files/20237302/icd_config_bin.zip)
This attachment include sdkconfig and merged bin, please refer to. The current idf version is v5.2.3