espressif/esp-matter #1389: ICD App using Matter over WiFi

TheBrightKnight, 2025-04-23:
Currently my device (ESP32C6) is using Matter over WiFi as it is also serving as a client for an external application sending REST calls. 

I have tried to implement the a light sleep (using the ICD example) but all the ICD configurations are disabled and cannot be changed (showing up in menuconfig in RED with "NEW" next to it). 

Is it because WiFi is enabled or something else? 
If the former, are there any plans to support it ? 

Currently I am using the native deep sleep command which is working well but it also needs to reconnect/recommission on every wake up which consumes a lot of power. 

[sdkconfig.txt](https://github.com/user-attachments/files/19863187/sdkconfig.txt)

Jerry-ESP, 2025-04-23:
The ICD is only work in thread device, unsupported in wifi device. If you want to decrease the  power consumption， you can simply enable the CONFIG_PM_ENABLE for wifi device.

TheBrightKnight, 2025-04-23:
> The ICD is only work in thread device, unsupported in wifi device. If you want to decrease the power consumption， you can simply enable the CONFIG_PM_ENABLE for wifi device.

Are there any benefits of using CONFIG_PM_ENABLE with deep sleep/light sleep native functions ? 
OR, what actually happens behind the scenes if enabled (without deep/light sleep) ?

Jerry-ESP, 2025-04-30:
Actually you have already enabled the CONFIG_PM_ENABLE in your sdkconfig, if you want to enable the light/deep sleep, you need to enable this option first.