espressif/esp-matter #1392: Build Failure on WSL

tanishqmanuja, 2025-04-24:
**Describe the bug**
Unable to build.

**Environment**
- ESP-Matter Commit Id: 5658ae930a2c757fcb83e39ad0bd99b21e30ffdb
- ESP-IDF Commit Id: 4c2820d377d1375e787bcef612f0c32c1427d183 (v5.4.1)
- SoC (eg: ESP32 or ESP32-C3): N/A
- Device Logs (Please attach the log file): N/A

- Host Machine OS: Ubuntu 22.04 WSL
- Host Machine Python version: Python 3.10.12

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
Logs: [full-install-logs.txt](https://github.com/user-attachments/files/19884938/full-install-logs.txt)

sayondeep, 2025-04-27:
> g++: fatal error: Killed signal terminated program cc1plus
> compilation terminated.
> [523/567] c++ obj/examples/chip-tool/commands/clusters/chip-tool-utils.ModelCommand.cpp.o
> ninja: build stopped: subcommand failed.

Your system is out of resources to continue the build.

Please use [--no-host-tool option ](https://github.com/espressif/esp-matter/blob/6ad504b596e51caea6368a85181b4a7e3f84585b/install.sh#L8) to install matter via `install.sh `

tanishqmanuja, 2025-04-27:
@sayondeep thanks, Increased Memory allocated to WSL instance and it worked 🥳

By default it only allocated 50% of the memory.

closed 2025-04-27 by tanishqmanuja