espressif/esp-matter #1395: Best way to allow for restoring state

tanishqmanuja, 2025-04-25:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

I cannot find in docs the suggested way to do this, can you please provide some context or resources.

I have checked all the examples from my side still confused

**Additional context**

Here what I was able to figure out, but this seems hacky
https://github.com/tanishqmanuja/matter.multi-plug-device/blob/14c4fa3cdce3619f3b22e5fa97ac5fa45c9d9bf8/main/app_driver.cpp#L136C1-L163C2

Also the solution should be able to support both Home Assistant and Other Hubs such as Alexa etc (Which doesn't show option to configure previous state restore)

[image]

**Other Similar Issues**
But the answer provided is use case based and not clear enough

https://github.com/espressif/esp-matter/issues/1194
https://github.com/espressif/esp-matter/issues/841

Jerry-ESP, 2025-04-25:
Hi @tanishqmanuja , Sorry, I don’t quite understand your detailed question. Could you please describe it further?

tanishqmanuja, 2025-04-25:
@Jerry-ESP , In detail with full context -- I wanted to make a matter enabled device comprising of 4channel relay and an esp32 ( this was long before examples/multiple_on_off_plugin_unit existed )

But I had to maintain 2 copies of vector array containing Gpio-endpoint mapping.

Then came the example i mentioned above, It had better code structure and didn't have to maintain 2 copies.

So I refactored my code just like the example. 
But .... Restore previous state from flash is not working for me in the multiple on off example.

### Main question here
How to implement restore from flash without doing hacky solutions like what i had to do here ..