espressif/esp-matter #1396: PICS-generator setup error

LinKvFM, 2025-04-27:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:02bbdb39eaca5d2329a64f6c779d7419031e9e03
- ESP-IDF Commit Id:c9763f62dd00c887a1a8fafe388db868a7e44069(tag v5.2.3)
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):

- Host Machine OS: ubuntu22.04
- Host Machine Python version: 3.10

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
See:  https://github.com/project-chip/connectedhomeip/issues/38624

Jerry-ESP, 2025-04-27:
You can try to use below command to install ***libgirepository1.0-dev*** 
```sudo apt install libgirepository1.0-dev```

LinKvFM, 2025-04-27:
Already installed it, but it still doesn't work

[image]

LinKvFM, 2025-04-27:
There is a folder called girepository-1.0 in the /lib directory.

[image]

Jerry-ESP, 2025-04-27:
试一下下面的几条命令：
sudo apt install libgirepository1.0-dev gir1.2-glib-2.0
sudo apt install libglib2.0-dev
pip install pygobject

LinKvFM, 2025-04-27:
```
sudo apt install libgirepository1.0-dev gir1.2-glib-2.0
sudo apt install libglib2.0-dev
pip install pygobject
```
还是不行。这几条包已经安装过了，并没有更新任何东西。

[image]

LinKvFM, 2025-04-27:
如果直接运行会出现： ModuleNotFoundError: No module named 'chip'

[image]

Jerry-ESP, 2025-04-27:
这个版本的PICS Tool配置起来还有一些问题，你可以切换到最新的master分支，使用最新的PICS generate tool