espressif/esp-matter #1397: Should have a concentration_measurement namespace

olavt, 2025-04-27:
In the Matter specification, all the Concentration Measurement clusters share the same attribute ids for MeasuredValue, AverageMeasuredValue and PeakMeasuredValue.

In the SDK these attribute ids are all defined under each specific namespace, so in the code one have to use a notation like this:

Pm10ConcentrationMeasurement::Attributes::MeasuredValue::Id
NitrogenDioxideConcentrationMeasurement::Attributes::MeasuredValue::Id
...

In my code I would like to avoid hard coding for each measured substance and I would like to update these attributes like this:

            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                0x00000000, // Measured Value
                matterAirQuality->m_measurements.GetLatest(clusterId));
    
            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                0x00000005, // Average Measured Value
                matterAirQuality->m_measurements.GetAverage(clusterId));

            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                0x00000003, // Peak Measured Value
                matterAirQuality->m_measurements.GetPeak(clusterId));

In order to avoid hard coding the attribute ids, it would make more sense to write:

            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                ConcentrationMeasurement::Attributes::MeasuredValue::Id,
                matterAirQuality->m_measurements.GetLatest(clusterId));
    
            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                ConcentrationMeasurement::Attributes::AverageMeasuredValue::Id,
                matterAirQuality->m_measurements.GetAverage(clusterId));

            UpdateAttributeValueFloat(
                airQualityEndpoint,
                clusterId,
                ConcentrationMeasurement::Attributes::PeakMeasuredValue::Id,
                matterAirQuality->m_measurements.GetPeak(clusterId));

So, I suggest adding the ConcentrationMeasurement namespace.

It would maybe have been better to have real c++ inheritance instead of namespaces.

jadhavrohit924, 2025-04-29:
Hi @olavt, this is a known issue. We will come up with something soon. Thanks for reporting.

pimpalemahesh, 2025-07-22:
@olavt The issue has been resolved with commit [afe082a](https://github.com/espressif/esp-matter/commit/afe082a1ffdd9e042d26ed3197bd012862465f88). Please close this issue!