espressif/esp-matter #1398: Unable to add device to Apple Home?

olavt, 2025-04-27:
I have create a Matter over Thread application, but when I try to add the device to Apple Home it doesn't succeed. I get the warning about adding an uncertified device and click "Add anyway", but then the process end with the error message "Unable to Add Accessory".

Attached is the log file from the Add Accessory process.

[AddAccessory.pdf](https://github.com/user-attachments/files/19931853/AddAccessory.pdf)

Jerry-ESP, 2025-04-29:
Did this issue happen every time? What is your IOS version and Homepod version? Was your DUT a test device?

olavt, 2025-04-29:
Yes, I have tried to add the device several time. I run iOS 18.4.1 on my phone and the same version on Apple TV. I'm not using HomePod. It's a test device created based on the light example. As I mentioned, I get the warning about adding an uncertified device. I have successfully been able to add test devices to Apple Home before.

Jerry-ESP, 2025-04-29:
> [@Jerry-ESP](https://github.com/Jerry-ESP) Yes, I have tried to add the device several times. I run iOS 18.4.1 on my phone and the same version on Apple TV. I'm not using HomePod. It's a test device created based on the light example. As I mentioned, I get the warning about adding an uncertified device. I have successfully been able to add test devices to Apple Home before.

It seems a issue occur after IOS 18.4.1 version update, the thread device can't be commissioning successfully. I think you can use an old ios version(18.3.2 is OK) or wait for the new version update .

olavt, 2025-04-29:
@Jerry-ESP Yes, that seems to be the case. I was able to add the device to my own test controller (using the Python Matter Server project).

jadhavrohit924, 2025-05-06:
@olavt Since it is an issue with iOS, can you please close the issue? Thanks!

olavt, 2025-05-13:
Working With iOS / TvOs 18.5.

closed 2025-05-13 by olavt