espressif/esp-matter #1399: Heap corruption during endpoint::destroy() method

DannyTheGhost, 2025-04-28:
**General description**
I am developing Matter bridge device with a UI, which allows to add and delete bridged devices to/from Matter. I noticed that there is a heap corruption that happens during removal of the bridged endpoints.
To eliminate possibilities that this is my own bug, I used 'zigbee-bridge' example, stripped it from Zigbee-related code, Source code is provided below with logs of the error.

**Environment**
- ESP-Matter Commit Id: checked on release/v1.4 (abbd0d2) and main (417f739) branches
- ESP-IDF Commit Id: present in v5.3 and v5.4.1
- SoC (eg: ESP32 or ESP32-C3): ESP-32WROVER-E module

- Host Machine OS: Ubuntu 22.04
- Host Machine Python version: v3.10.12

**Details**
Test application is based on 'zigbee-bridge' example project. Modifications:
- Zigbee-related code is removed
- After Matter core is initialized, thread for testing is created.
- Testing thread would create after delay some device endpoints, and then would attempt to delete one of them.
- Thread will dump heap information and check for integrity before endpoint deletion 

Stack trace after crash:

> (31041) chip[ZCL]: Shuting down on/off server cluster on endpoint 16[0m
> 
> CORRUPT HEAP: Bad head at 0x3fff3164. Expected 0xabba1234 got 0xfefefefe
> 
> assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)
> 
> 
> Backtrace: 0x40081c69:0x3fffb220 0x400912a5:0x3fffb240 0x400989d9:0x3fffb260 0x4009649b:0x3fffb380 0x400827b1:0x3fffb3a0 0x400825bb:0x3fffb3c0 0x4008273d:0x3fffb400 0x40098a09:0x3fffb420 0x40084089:0x3fffb440 0x400db769:0x3fffb460 0x401025ae:0x3fffb490 0x401020de:0x3fffb4b0 0x400d8acd:0x3fffb510 0x40091ce6:0x3fffb550
> 0x40081c69: panic_abort at /home/denys/_WORK/ESP/SDK/esp-idf/components/esp_system/panic.c:454
> 0x400912a5: esp_system_abort at /home/denys/_WORK/ESP/SDK/esp-idf/components/esp_system/port/esp_system_chip.c:92
> 0x400989d9: __assert_func at /home/denys/_WORK/ESP/SDK/esp-idf/components/newlib/assert.c:80
> 0x4009649b: multi_heap_free at /home/denys/_WORK/ESP/SDK/esp-idf/components/heap/multi_heap_poisoning.c:279 (discriminator 1)
> 0x400827b1: heap_caps_free at /home/denys/_WORK/ESP/SDK/esp-idf/components/heap/heap_caps_base.c:70
> 0x400825bb: trace_free at /home/denys/_WORK/ESP/SDK/esp-idf/components/heap/include/heap_trace.inc:164
> 0x4008273d: __wrap_heap_caps_free at /home/denys/_WORK/ESP/SDK/esp-idf/components/heap/include/heap_trace.inc:168
> 0x40098a09: free at /home/denys/_WORK/ESP/SDK/esp-idf/components/newlib/heap.c:39
> 0x40084089: esp_matter_mem_free(void*) at /home/denys/_WORK/ESP/SDK/esp-matter/components/esp_matter/utils/esp_matter_mem.cpp:47
> 0x400db769: esp_matter::endpoint::destroy(unsigned int*, unsigned int*) at /home/denys/_WORK/ESP/SDK/esp-matter/components/esp_matter/esp_matter_core.cpp:1810
> 0x401025ae: esp_matter_bridge::remove_device(esp_matter_bridge::device*) at /home/denys/_WORK/ESP/SDK/esp-matter/components/esp_matter_bridge/esp_matter_bridge.cpp:392
> 0x401020de: app_bridge_remove_device(app_bridged_device*) at /home/denys/_WORK/ESP/SDK/esp-matter/examples/common/app_bridge/app_bridged_device.cpp:279
> 0x400d8acd: virtualBridgeTask(void*) at /home/denys/_WORK/ESP/projects/zigbee_bridge/main/app_main.cpp:227 (discriminator 1)
> 0x40091ce6: vPortTaskWrapper at /home/denys/_WORK/ESP/SDK/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Logs and project files are included below
[log.zigbee_bridge.20250428120019.txt](https://github.com/user-attachments/files/19937566/log.zigbee_bridge.20250428120019.txt)

[test_project.zip](https://github.com/user-attachments/files/19937594/test_project.zip)

pimpalemahesh, 2025-05-02:
@DannyTheGhost I’ve been able to reproduce the issue on my end. It appears to be caused by a double free of the same memory allocation. We’ll work on a fix shortly. Thanks for reporting it!

pimpalemahesh, 2025-05-09:
@DannyTheGhost Could you please remove [this line](https://github.com/espressif/esp-matter/blob/74caefc7fb2e048685e55f43db55744aa3fd0c3c/components/esp_matter/esp_matter_core.cpp#L1830)? It's causing memory poisoning due to a double free of attribute metadata.

DannyTheGhost, 2025-05-12:
@pimpalemahesh It seems that Your solution helped, and I am no longer observing heap corruption. Further testing is required, as I am not proficient enough to say that there is no any other memory poisoning related to endpoint::destroy(). 
Anyway, Thank You for the solution!

pimpalemahesh, 2025-05-15:
@DannyTheGhost This change has been merged and is now available in the latest main branch. You may close this issue.

closed 2025-05-15 by dhrishi