espressif/esp-matter #1402: Can ESP32-C6 run ESP-NOW communication in parallel with Matter over Thread?

bilalahmaddev, 2025-04-28:
I am exploring a design where the ESP32-C6 device will use:

- ESP-NOW for fast local peer-to-peer communication, and
- Matter over Thread for smart home integration.

Since ESP32-C6 supports both Wi-Fi (for ESP-NOW) and IEEE 802.15.4 (for Thread) radios, I would like to confirm:

- Is it supported to use ESP-NOW and Thread simultaneously on the ESP32-C6?
- Are there any coexistence considerations, limitations, or API requirements when running both protocols in parallel?
- Is there a recommended way to manage resource conflicts or prioritize traffic between Wi-Fi (ESP-NOW) and Thread (Matter) in ESP-IDF/ESP-Matter?

wqx6, 2025-06-12:
Hi, we didn't test the co-exist scenario for ESP-NOW and IEEE802.15.4. But it should be supported to use ESP-NOW and Thread simultaneously on the ESP32-C6 as C6 can use Wi-Fi and Thread simultaneously.

> Are there any coexistence considerations, limitations, or API requirements when running both protocols in parallel?

Yes, since there is only one radio on ESP32-C6, so the Wi-Fi and Thread can not receive packets at the same time, which will result in packet loss. We would suggest that the C6 should work as a Thread End-Device instead of Thread Router. No extra API should be called as the co-exist is integrated with ESP-IDF.

>  Is there a recommended way to manage resource conflicts or prioritize traffic between Wi-Fi (ESP-NOW) and Thread (Matter) in ESP-IDF/ESP-Matter?

Please use C6 as an MTD(Minimal Thread device ).

bilalahmaddev, 2025-06-12:
@wqx6 so you do not recommend using both Thread and esp-now on esp32-c6?