espressif/esp-matter #1405: Solar Energy Device: API Support for Setting Semantic Tags in ESP-Matter

bilalahmaddev, 2025-05-02:
I'm working on a solar energy device setup using ESP-Matter v1.4 with a parent SolarPower device and multiple child ElectricalEnergyMeasurement devices. According to the Matter Device Library Specification R1.4, I need to establish a semantic relationship (e.g., isPartOf) between the children and the parent device using the Semantic Tags cluster and Descriptor cluster.

Is there an official API available in ESP-Matter to set semantic tags (e.g., TagList attribute) programmatically during endpoint initialization?

If not, could you guide me on the correct approach to encode and write semantic tags via the current attribute APIs?

Thank you!

lboue, 2025-05-02:
Hi @bilalahmaddev,

Maybe you can look at the generic_switch examples here:

https://github.com/espressif/esp-matter/blob/78bc8e21c932a5014a08ecdd8bfa5535a771c107/examples/generic_switch/main/app_main.cpp#L242-L243

And here in the doc:
https://github.com/espressif/esp-matter/blob/78bc8e21c932a5014a08ecdd8bfa5535a771c107/docs/en/developing.rst#2524-features

bilalahmaddev, 2025-05-02:
@lboue I have use it before but I am not sure how to relate other electrical sensor with a parent solar power device while defining these for solar power device:

```
namespace {
// Please refer to https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/namespaces
constexpr const uint8_t kNamespaceSwitches = 43;
// Common Number Namespace: 7, tag 0 (Zero)
constexpr const uint8_t kTagSwitchOn = 0;
// Common Number Namespace: 7, tag 1 (One)
constexpr const uint8_t kTagSwitchOff = 1;

constexpr const uint8_t kNamespacePosition = 8;
// Common Position Namespace: 8, tag: 0 (Left)
constexpr const uint8_t kTagPositionLeft = 0;
// Common Position Namespace: 8, tag: 1 (Right)
constexpr const uint8_t kTagPositionRight = 1;

const Descriptor::Structs::SemanticTagStruct::Type gEp1TagList[] = {
    {.namespaceID = kNamespaceSwitches, .tag = kTagSwitchOn},
    {.namespaceID = kNamespacePosition, .tag = kTagPositionRight}};
const Descriptor::Structs::SemanticTagStruct::Type gEp2TagList[] = {
    {.namespaceID = kNamespaceSwitches, .tag = kTagSwitchOff},
    {.namespaceID = kNamespacePosition, .tag = kTagPositionLeft}};

}

```

lboue, 2025-05-02:
You probably have to set the parent endpoint Descriptor cluster `PartsList` attribute.

lboue, 2025-05-03:
I found an example here:

**Refrigerator example**
https://github.com/espressif/esp-matter/blob/78bc8e21c932a5014a08ecdd8bfa5535a771c107/examples/refrigerator/main/app_main.cpp#L129

So you have to call `set_parent_endpoint` to set the `parts_list` attribute for the parent, when the parent is a composite endpoint. 

https://github.com/espressif/esp-matter/blob/78bc8e21c932a5014a08ecdd8bfa5535a771c107/components/esp_matter/esp_matter_core.h#L327-L338

bilalahmaddev, 2025-05-03:
@lboue Thanks for sharing. What do you think about this approach:
```
  // Define child endpoints in PartsList (as uint16_t array)
    uint16_t parts_list[] = { 1, 2 };
    uint16_t parts_list_count = sizeof(parts_list) / sizeof(uint16_t);

    // Convert to uint8_t* (required by API)
    uint8_t *parts_list_bytes = (uint8_t *)parts_list;

    // Create the PartsList attribute
    esp_matter::cluster::descriptor::attribute::create_parts_list(
        descriptor,
        parts_list_bytes,           // Raw bytes of the parts_list array
        sizeof(parts_list),         // Total bytes (2 bytes per endpoint ID)
        parts_list_count            // Number of endpoints in the list
    );
```

lboue, 2025-05-04:
Do it work?

I think you should also add device types to your main `SolarPower` EP:

```c++
/* DEVICE_ENERGY_MANAGEMENT */
add_device_type(endpoint, ESP_MATTER_DEVICE_ENERGY_MANAGEMENT_DEVICE_TYPE_ID, ESP_MATTER_DEVICE_ENERGY_MANAGEMENT_DEVICE_TYPE_VERSION);

/* POWER_SOURCE */
add_device_type(endpoint, ESP_MATTER_POWER_SOURCE_DEVICE_TYPE_ID, ESP_MATTER_POWER_SOURCE_DEVICE_TYPE_VERSION);

/* ELECTRICAL_SENSOR */
add_device_type(endpoint, ESP_MATTER_ELECTRICAL_SENSOR_DEVICE_TYPE_ID, ESP_MATTER_ELECTRICAL_SENSOR_DEVICE_TYPE_VERSION);
```

bilalahmaddev, 2025-05-04:
I think this is not required, solar power endpoint already have this.

lboue, 2025-05-14:
> [@lboue](https://github.com/lboue) Thanks for sharing. What do you think about this approach:
> 
> ```
>   // Define child endpoints in PartsList (as uint16_t array)
>     uint16_t parts_list[] = { 1, 2 };
>     uint16_t parts_list_count = sizeof(parts_list) / sizeof(uint16_t);
> 
>     // Convert to uint8_t* (required by API)
>     uint8_t *parts_list_bytes = (uint8_t *)parts_list;
>     
>    cluster_t* descriptor = cluster::get(endpoint, Descriptor::Id); // parent
>     // Create the PartsList attribute
>     esp_matter::cluster::descriptor::attribute::create_parts_list(
>         descriptor,
>         parts_list_bytes,           // Raw bytes of the parts_list array
>         sizeof(parts_list),         // Total bytes (2 bytes per endpoint ID)
>         parts_list_count            // Number of endpoints in the list
>     );
> ```

Does it work?

jadhavrohit924, 2025-06-17:
What is the issue here? If the issue is resolved please close it.

closed 2025-06-17 by bilalahmaddev