espressif/esp-matter #1407: Matter example with energy reporting capabilities

lboue, 2025-05-04:
**Is your feature request related to a problem? Please describe.**
I would like a Matter example with energy reporting capabilities

**Describe the solution you'd like**
A Matter example with energy reporting capabilities

**Describe alternatives you've considered**
Facing using building it myself

**Additional context**
Using Electrical sensor.

Jerry-ESP, 2025-05-06:
There isn't an example for the Electrical sensor, Please refer to the implementation  this issue: https://github.com/espressif/esp-matter/issues/1102#issuecomment-2408704757

lboue, 2025-05-06:
Hi,

I did read this thread. I've created an [Solar_Power example](https://github.com/lboue/esp-matter/blob/Solar_Power/examples/solar_power/main/app_main.cpp) that works to udapte following attributes:
- ElectricalPowerMeasurement::Attributes::Voltage
- ElectricalPowerMeasurement::Attributes::ActiveCurrent
- ElectricalPowerMeasurement::Attributes::ActivePower
    

But I am facing an issue implementing code from this post: https://github.com/espressif/esp-matter/issues/1102#issuecomment-2408704757

Here is my test for ElectricalEnergyMeasurement::SetMeasurementAccuracy:
https://github.com/lboue/esp-matter/blob/e1080409e88ba3003afcc0bd747f5304f42b2bfb/examples/solar_power/main/app_main.cpp#L197-L228

When I enable this line, I am facing a crash:
https://github.com/lboue/esp-matter/blob/e1080409e88ba3003afcc0bd747f5304f42b2bfb/examples/solar_power/main/app_main.cpp#L231

Error log:
https://github.com/lboue/esp-matter/blob/Solar_Power/examples/solar_power/error-2025-05-06.log

Did I miss something?

Regards
Ludovic

jadhavrohit924, 2025-05-06:
@lboue As the crash suggests, code is not safe. Use `chip_stack_lock` and `chip_stack_unlock` before and after `ElectricalEnergyMeasurement::SetMeasurementAccuracy`

lboue, 2025-05-06:
I am facing another issue
https://github.com/espressif/esp-matter/blob/78bc8e21c932a5014a08ecdd8bfa5535a771c107/components/esp_matter/esp_matter_endpoint.cpp#L1902

So it should enable those features:
- exported_energy
- cumulative_energy

But I am getting another value using chip-tool: 5
```
./chip-tool electricalenergymeasurement read feature-map 100 1 | grep DMG
[1746530455.529] [505689:505689] [DMG] Ember attribute persistence requires setting up
[1746530458.938] [505689:505692] [DMG] SendReadRequest ReadClient[0x741100010940]: Sending Read Request
[1746530458.938] [505689:505692] [DMG] 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
[1746530458.939] [505689:505692] [DMG] MoveToState ReadClient[0x741100010940]: Moving to [AwaitingIn]
[1746530459.046] [505689:505692] [DMG] ReportDataMessage =
[1746530459.046] [505689:505692] [DMG] {
[1746530459.046] [505689:505692] [DMG] 	AttributeReportIBs =
[1746530459.046] [505689:505692] [DMG] 	[
[1746530459.046] [505689:505692] [DMG] 		AttributeReportIB =
[1746530459.046] [505689:505692] [DMG] 		{
[1746530459.046] [505689:505692] [DMG] 			AttributeDataIB =
[1746530459.046] [505689:505692] [DMG] 			{
[1746530459.046] [505689:505692] [DMG] 				DataVersion = 0xc528dd1c,
[1746530459.046] [505689:505692] [DMG] 				AttributePathIB =
[1746530459.046] [505689:505692] [DMG] 				{
[1746530459.046] [505689:505692] [DMG] 					Endpoint = 0x1,
[1746530459.046] [505689:505692] [DMG] 					Cluster = 0x91,
[1746530459.046] [505689:505692] [DMG] 					Attribute = 0x0000_FFFC,
[1746530459.046] [505689:505692] [DMG] 				}
[1746530459.046] [505689:505692] [DMG] 					
[1746530459.046] [505689:505692] [DMG] 				Data = 5 (unsigned), 
[1746530459.046] [505689:505692] [DMG] 			},
[1746530459.046] [505689:505692] [DMG] 			
[1746530459.046] [505689:505692] [DMG] 		},
[1746530459.046] [505689:505692] [DMG] 		
[1746530459.046] [505689:505692] [DMG] 	],
[1746530459.046] [505689:505692] [DMG] 	
[1746530459.046] [505689:505692] [DMG] 	SuppressResponse = true, 
[1746530459.046] [505689:505692] [DMG] 	InteractionModelRevision = 12
[1746530459.046] [505689:505692] [DMG] }
```

And in the logs this:
```
E (322569) chip[DMG]: Attribute type 0x48 not handled
E (322589) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_009C, attributeId: 0x0000_0000err = 501
I (322599) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x0000FFFC is 3 **********
I (322609) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000000 is 0 **********
I (322619) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000001 is 3 **********
I (322629) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000002 is <invalid type: 4> **********
E (322639) chip[DMG]: Attribute type 0x48 not handled
E (322639) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0090, attributeId: 0x0000_0002err = 501
I (322659) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000008 is 20000 **********
I (322669) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x0000FFFD is 1 **********
I (322679) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000004 is 230000 **********
I (322689) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x00000005 is 2000 **********
I (322699) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000091's Attribute 0x0000FFFD is 1 **********
E (322709) chip[ZCL]: Electrical Energy Measurement: can not get CumulativeEnergyExported, feature is not supported
E (322729) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0091, attributeId: 0x0000_0002err = 6c
```

Is this an issue in esp-matter source code?

Regards

lboue, 2025-05-07:
I  found my mistake here:
```c++
	auto mask = chip::BitMask<chip::app::Clusters::ElectricalEnergyMeasurement::Feature>(
        chip::app::Clusters::ElectricalEnergyMeasurement::Feature::kCumulativeEnergy, 
        chip::app::Clusters::ElectricalEnergyMeasurement::Feature::kImportedEnergy);
```

lboue, 2025-05-09:
Now my ElectricalPowerMeasurementDelegate is working and `Accuracy` attribute value is OK
```
    solar_power::config_t solar_power_config;
    static chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate epm_delegate;
    solar_power_config.electrical_sensor.electrical_power_measurement.delegate = &epm_delegate;
```

But now Voltage and ActiveCurrent attributes ok NOK:
```
 (32439) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x0000FFFD is 1 **********
E (32449) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0090, attributeId: 0x0000_0004err = 586
E (32459) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0090, attributeId: 0x0000_0005err = 586
I (32479) chip[EM]: <<< [E:28637r S:54525 M:25444143 (Ack:268435274)] (S) Msg TX from 0000000000000012 to 3:000000000001B669 [5A16] [UDP:[FE80::B63C:CA04:ABE:3AC0%st1]:32901] --- Type 0001:05 (IM:ReportData) (B:1187)
I (32489) chip[EM]: ??1 [E:28637r S:54525 M:25444143] (S) Msg Retransmission to 3:000000000001B669 scheduled for 333ms from now [State:Active II:500 AI:300 AT:4000]
I (32509) chip[EM]: >>> [E:28637r S:54525 M:268435275 (Ack:25444143)] (S) Msg RX from 3:000000000001B669 [5A16] to 0000000000000012 --- Type 0001:01 (IM:StatusResponse) (B:42)
```

How can I use the delegate to update those attributes values?

jadhavrohit924, 2025-06-17:
The example is out in all-device-types app please check the latest commit.