espressif/esp-matter #1410: How to properly use attribute_cb of subscribe_command() ?

dmartauz, 2025-05-07:
Could you please provide example how to properly use attribute_cb of subscribe_command()?
https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_subscribe_command.h

wqx6, 2025-05-08:
Please refer to this doc https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#subscribe-commands

dmartauz, 2025-05-08:
@wqx6 are new attribute data passed to the callback as a parameter?

wqx6, 2025-05-08:
> [@wqx6](https://github.com/wqx6) are new attribute data passed to the callback as a parameter?

Yes, the attribute data will be passed to the attribute_report_cb in the TLVReader.

dhrishi, 2025-05-14:
@dmartauz Please close the issue if your question is answered

closed 2025-05-19 by dmartauz