espressif/esp-matter #1411: How to set `ElectricalPowerMeasurement` Cluster `Accuracy` attribute

lboue, 2025-05-07:
**Describe the question/query that you have**
How to set `ElectricalPowerMeasurement` Cluster `Accuracy` attribute.

**Additional context**

I have a List[ElectricalPowerMeasurement.Structs.MeasurementAccuracyStruct] in my code.
Could you tell me how to set `ElectricalPowerMeasurement` Cluster `Accuracy` attribute.

Regards

wqx6, 2025-05-08:
For ElectricalPowerMeasurement cluster, the accuracy attribute is managed internally by the attribute access interface (see https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/electrical-power-measurement-server/electrical-power-measurement-server.cpp#L80). And there is a delegate which maintains the accuracy list so you should implement a delegate for that cluster so that you can set the related attributes for that cluster.

lboue, 2025-05-08:
Is there an example of delegate implementation in the ESP-Matter SDK?

crivi30, 2025-05-08:
I'm also interested in this, and I'd like to add that the Accuracy attribute returns an error when I try to read it using chip-tool.

Here is chip-tool log:
```

[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFFC DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   FeatureMap: 3
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_0000 DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   PowerMode: 0
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_0001 DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   NumberOfMeasurementTypes: 0

**[1746695634.125] [198924:198965] [TOO] Response Failure: IM Error 0x00000501: General error: 0x01 (FAILURE)**

[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_0008 DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   ActivePower: 0
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFFD DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   ClusterRevision: 1
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFF8 DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   GeneratedCommandList: 0 entries
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFF9 DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   AcceptedCommandList: 0 entries
[1746695634.125] [198924:198965] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFFB DataVersion: 2565214521
[1746695634.125] [198924:198965] [TOO]   AttributeList: 9 entries
[1746695634.125] [198924:198965] [TOO]     [1]: 65528 (GeneratedCommandList)
[1746695634.125] [198924:198965] [TOO]     [2]: 65529 (AcceptedCommandList)
[1746695634.125] [198924:198965] [TOO]     [3]: 65531 (AttributeList)
[1746695634.125] [198924:198965] [TOO]     [4]: 65532 (FeatureMap)
[1746695634.125] [198924:198965] [TOO]     [5]: 0 (PowerMode)
[1746695634.125] [198924:198965] [TOO]     [6]: 1 (NumberOfMeasurementTypes)
[1746695634.125] [198924:198965] [TOO]     [7]: 2 (Accuracy)
[1746695634.125] [198924:198965] [TOO]     [8]: 8 (ActivePower)
[1746695634.125] [198924:198965] [TOO]     [9]: 65533 (ClusterRevision)
```

It appears that reading it triggers a general failure (IM Error 0x00000501). Do you know which could be the issue?

wqx6, 2025-05-08:
Please refer to this doc https://docs.espressif.com/projects/esp-matter/en/latest/esp32/app_guide.html?highlight=delegate#delegate-implementation

lboue, 2025-05-08:
So in our case that would be it, wouldn't it?
```c++
#include "app/clusters/electrical-power-measurement-server/electrical-power-measurement-server.h"

solar_power::config_t solar_power_config;
static chip::app::Clusters::ElectricalPowerMeasurement::Delegate epm_delegate;
solar_power_config.delegate = epm_delegate;
endpoint_t *endpoint = solar_power::create(node, &solar_power_config, ENDPOINT_FLAG_NONE, NULL);
```

lboue, 2025-05-08:
So in our case that would be it, wouldn't it?
```c++
#include "app/clusters/electrical-power-measurement-server/electrical-power-measurement-server.h"

  solar_power::config_t solar_power_config;
  static chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate epm_delegate;
  solar_power_config.electrical_sensor.electrical_power_measurement.delegate = &epm_delegate;
  
  endpoint_t *endpoint = solar_power::create(node, &solar_power_config, ENDPOINT_FLAG_NONE, NULL);
  ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create solar power endpoint"));
```

But I am getting an error:
```c++

home/ludovic/esp-matter/examples/solar_power/main/app_main.cpp: In function 'void app_main()':
/home/ludovic/esp-matter/examples/solar_power/main/app_main.cpp:118:61: error: 'ElectricalPowerMeasurementDelegate' in namespace 'chip::app::Clusters::ElectricalPowerMeasurement' does not name a type
  118 |     static chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate epm_delegate;
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ludovic/esp-matter/examples/solar_power/main/app_main.cpp:119:83: error: 'epm_delegate' was not declared in this scope
  119 |     solar_power_config.electrical_sensor.electrical_power_measurement.delegate = &epm_delegate;
      |                                                                                   ^~~~~~~~~~~~
```

wqx6, 2025-05-09:
Did you include the header [ElectricalPowerMeasurementDelegate.h](https://github.com/espressif/connectedhomeip/blob/ea679d2dc674f576f4d391d1d71af1489010e580/examples/energy-management-app/energy-management-common/energy-reporting/include/ElectricalPowerMeasurementDelegate.h) in app_main.cpp? Note that the ElectricalPowerMeasurementDelegate is only an example implementation for the cluster delegate, you can realize your inheritance from the virtual class in which your can set the accuracy attribute for the cluster.

lboue, 2025-05-09:
Thanks for the info, I've corrected that and have another error. Do you know what the problem is this time?
```
/home/ludovic/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(app_main.cpp.obj): in function `app_main':
/home/ludovic/esp-matter/examples/solar_power/main/app_main.cpp:119:(.text.app_main+0x136): undefined reference to `vtable for chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate'
/home/ludovic/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /home/ludovic/esp-matter/examples/solar_power/main/app_main.cpp:119:(.text.app_main+0x142): undefined reference to `vtable for chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate'
collect2: error: ld returned 1 exit status
```

Full log: [vtable_err.log](https://github.com/lboue/esp-matter/blob/9c553dd4b1505f62f3892880844a13f1dce60f5e/examples/solar_power/vtable_err.log)

lboue, 2025-05-09:
I had also to add `ElectricalPowerMeasurementDelegate.cpp ` in [CMakeLists.txt](https://github.com/lboue/esp-matter/commit/3662773a01b7e4e7ac97d31b7015a1a371fbb742).
Now it builds.

[image]

closed 2025-05-09 by lboue