espressif/esp-matter #1413: I2C driver conflict: driver_ng is not allowed to be used with old driver (ESP-IDF 5.4 / ESP32-H2)

donaldkohcw, 2025-05-09:
Hi Espressif, 

I'm encountering an I2C driver conflict error when running the ESP-Matter "sensor" example using ESP-IDF v5.4.1 on an ESP32-H2. It built without issue. The error occurs during boot, shortly after SPI flash initialization.

"I (918) spi_flash: flash io: dio
E (921) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
abort() was called at PC 0x420c8ff7 on core 0
--- 0x420c8ff7: check_i2c_driver_conflict at .../components/driver/i2c/i2c.c:1723"

I could not find any option in `menuconfig` to explicitly enable or disable the NG driver. Is there a known workaround or method to confirm that only the NG driver is being used? I would appreciate any guidance.

Best regards, 
DK

dk77148, 2025-05-27:
Hi Espressif Team,

Just following up on this question—wondering if there are any updates or suggestions regarding the I2C driver conflict when running the sensor example on the ESP32-H2 with ESP-IDF v5.4.1. Appreciate your support!

Thanks again.

Best regards,
DK

crushonme, 2025-05-29:
Update your application to user i2c ng API.

- Use i2c_master_bus_config_t instead of i2c_config_t
- Use i2c_new_master_bus to replace i2c_param_config and i2c_driver_install.

For details, you can refer to esp32-camera sccb.c and sccb-ng.c

dpnebert, 2025-06-13:
Hi, I am unable to find any references to the old I2C driver.  The header "i2c.h" is not found in my project at all.

I love the components, but this is the first time that components being so "robust" has screwed me over.  I went through all the managed_components and delete all the demo, example, test directory, and when I run the PowerShell command from my project root:
"
Get-ChildItem -Recurse | Select-String "i2c_param_config"
"

Nothing is found.  When I search for:
"
Get-ChildItem -Recurse | Select-String "i2c_config_t"
"

I get results:
"
main\ch350hv37a-ct.c:233:    const esp_lcd_panel_io_i2c_config_t tp_io_config = ESP_LCD_TOUCH_IO_I2C_FT6X36_CONFIG();
main\ch350hv37a-ct.c:244://     const esp_lcd_panel_io_i2c_config_t tp_io_config = ESP_LCD_TOUCH_IO_I2C_FT6X36_CONFIG();
"

As you can tell, I'm working on the touch screen part of the CH350HV37A-CT lcd display.

Instead of tell me how to find where the legacy driver is being used, can we update the IDF to atleast tell me the offending file to look into?

It reminds me when a method fails because one of three handles are null.  It will tell you that you can't use a null handle, but doesn't tell you which one.  I've had to go in to the source and change it so it will fail separately for each offending NULL handle.