espressif/esp-matter #1414: The device type recognition of Apple, Google, and Amazon apps is incorrect

MaplestoryAlen, 2025-05-09:
I added two device clusters on the thermostat (ID 0x301), one for the fan and one for the humidity sensor. However, the displayed interface is completely different. Sometimes it shows the thermostat, one for the fan, or one for the humidity sensor. I have added them on the Apple Home app, Google app, and Amazon Alexa app. Why does the correct UI display the thermostat instead of the fan/humidity sensor? May I ask where I went wrong?

code:

       node::config_t node_config;
       node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

       thermostat::config_t thermostat_config;
	endpoint_t *endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);

	fan::config_t fan_config;
	endpoint_t *endpoint_1 = fan::create(node, &fan_config, ENDPOINT_FLAG_NONE, NULL);

	humidity_sensor::config_t humidity_config;
	endpoint_t *endpoint_2 = humidity_sensor::create(node, &humidity_config, ENDPOINT_FLAG_NONE, NULL);

	
    /* These node and endpoint handles can be used to create/add other endpoints and clusters.  */
    if (!node || !endpoint || !endpoint_1 ||!endpoint_2) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }
    thermostat_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "thermostat created with endpoint_id %d", thermostat_endpoint_id);	
	
	fan_endpoint_id = endpoint::get_id(endpoint_1);
	ESP_LOGI(TAG, "fan created with endpoint_id %d", fan_endpoint_id);	

	humidity_endpoint_id = endpoint::get_id(endpoint_2);
	ESP_LOGI(TAG, "humidity created with endpoint_id %d", humidity_endpoint_id);	
	
	err = set_parent_endpoint(endpoint_1, endpoint);
	if (err != ESP_OK) 
	{
	  ESP_LOGE(TAG, "Failed to set parent %d", err);  
	}
	err = set_parent_endpoint(endpoint_2, endpoint);
	if (err != ESP_OK) 
	{
	  ESP_LOGE(TAG, "Failed to set parent %d", err);
	}

Jerry-ESP, 2025-05-20:
Please share your test result, what do their displays look like in different app?

MaplestoryAlen, 2025-05-23:
Why does the UI display a temperature and humidity sensor and two temperature controllers when adding the same type of device to the Apple Home app?
[image]

[image]

> Please share your test result, what do their displays look like in different app? You created three endpoints, means three different devices. In different apps, may have different displays.

Jerry-ESP, 2025-05-26:
For a device, maybe have different displays in different ecosystem, it depends on their implementation, there isn't a standard for this point.  And for the second question: I'm not sure, it seems that Apple only show one temperature/humidity sensor in a room .