espressif/esp-matter #1418: Bridged rocker style switch

mikesolsen, 2025-05-15:
Hello,
I'm implementing a matter bridge, and I'm so far not able to add support for a rocker style switch with 2 buttons (on & off, not two separate rockers).

I've experimented with a couple approaches:
1) Using 2x bridged generic_switch endpoints per device (one for off, one for on). This works in Apple Home, however the on and off button are presented as separate devices in Apple Home.  Less than ideal.
2) Using a bridged generic_switch endpoint, and configuring the switch cluster to be multi position. I haven't seen this work.
3) Using endpoint composition: A bridged_node with two generic_switch children. I haven't seen this work.

Has anybody succeeded at bridging this style of switch?  How can I know if Apple/Google/Amazon support it? 

The comment here suggests it's not possible, but I'm hoping the comment is outdated or not discussing my situation: https://github.com/espressif/esp-matter/issues/1121#issuecomment-2448922409

Happy to share code if somebody can confirm which approach is actually possible with Apple/Google/Amazon.

Thanks!
Mike

jonsmirl, 2025-06-01:
You have a two position momentary contact switch, not two switches. So option #2 is correct.  Apple Home does not care if it is bridged or not, both function identically. That's how it should work in Matter, but of course Apple does not implement all of Matter. This works fine from the Home Kit APIs. But, last I checked, Apple Home Automation didn't support anything except a single position push button, it just ignores the position parameter in the event.

So based on my current understanding of Apple Home's automations, you can't make this work like you want it to.

mikesolsen, 2025-06-02:
Thanks Jon. Your previous comment had me suspecting as much.