espressif/esp-matter #1420: General Diagnostic Cluster issue: TestEventTriggersEnabled always FALSE

crivi30, 2025-05-16:
I’m having an issue enabling the **Test Event Trigger** of General Diagnostic cluster for **Test Harness certification**.

When the cluster is created by ESP Matter, the default value for the attribute is `false`:

```cpp
attribute::create_test_event_triggers_enabled(cluster, 0);
```

The flag is `ATTRIBUTE_FLAG_MANAGED_INTERNALLY`, so I believe the attribute is not writable.

I looked for a static configuration parameter to set before building, and found the following in `Kconfig`:

```kconfig
config TEST_EVENT_TRIGGER_ENABLED
    bool "Enable Test Event Trigger"
    default y
    help
        Enable TestEventTrigger in the GeneralDiagnostics cluster.

config TEST_EVENT_TRIGGER_ENABLE_KEY
    string "Test Event Trigger Enable Key"
    depends on TEST_EVENT_TRIGGER_ENABLED
    default "00112233445566778899aabbccddeeff"
    help
        The EnableKey in hex string format used by the TestEventTrigger command in the GeneralDiagnostics
        cluster. The string should be 32 characters long.
```

This should mean that the default value of the attribute is `true`. I also double-checked in my IDF `menuconfig`, and it’s marked as **Enabled**.

However, when I try to read the attribute using `chip-tool`, I get:

```
[1747397914.716] [249272:249304] [TOO] Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_0008 DataVersion: 3262397771
[1747397914.716] [249272:249304] [TOO]   TestEventTriggersEnabled: FALSE
```

Even when I try to set or update the value using the Matter SDK, it returns a failure.

My question is, How can I enable the Test Event Trigger for the Test Harness?
Thank you in advance

VaishaliAvhale, 2025-05-17:
Please tell me which test case you are executing and for which matter release.

crivi30, 2025-05-18:
Hi @VaishaliAvhale 
It is the test TC-EPM-2.2 for Electrical Power Measurement Cluster
The release is Matter 1.4
Thank you!

Jerry-ESP, 2025-05-19:
Hi @crivi30 ,

For the test event trigger related test cases, you should set the test event delegate through this function: https://github.com/espressif/esp-matter/blob/057ac6a3e06f29a8a6f0e942228d01064e6942d3/components/esp_matter/esp_matter_test_event_trigger.cpp#L23, then it will enable the TestEventTriggersEnabled attribute. 

For the delegate definition, you can refer to : https://github.com/project-chip/connectedhomeip/blob/ba6168ccd9f485f6d41a395b65c294ada203b9ea/examples/platform/esp32/common/Esp32AppServer.cpp#L120-L150  and https://github.com/project-chip/connectedhomeip/blob/ba6168ccd9f485f6d41a395b65c294ada203b9ea/examples/energy-management-app/energy-management-common/energy-evse/src/EnergyEvseEventTriggers.cpp#L140-L199 .

Please try to add the delegate, if still have any questions, please let me know

crivi30, 2025-05-20:
I will try! Thanks

crivi30, 2025-05-21:
Hi @Jerry-ESP,  
I followed your suggestion and I can now see `TestEventTriggersEnabled: TRUE`.  

However, I'm currently blocked at step 3 of test **TC-EPM-2.2**. I even tried using the `chip-tool` outside of the TH with the following command:

`chip-tool generaldiagnostics test-event-trigger 0xcafebabedeadbeef1122334455667788 0x009100000000000 1 0`

Here, `0xcafebabedeadbeef1122334455667788` is the enable key I set in `menuconfig`, and `0x009100000000000` is the event key.  

But I keep receiving the following error:  

`[1747822686.636] [191476:191476] [TOO] Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)`

Could you confirm the correct way to trigger this event?  
Also, in the case of TH, how can I add the PIXIT for all the steps? I'm currently trying with:

```
"test_parameters": {
  "endpoint": 1,
  "string-arg": "PIXIT.EPM.TEST_EVENT_TRIGGER:0xcafebabedeadbeef1122334455667788"
}
```

Thanks in advance!

Jerry-ESP, 2025-05-29:
What steps have you done so far? Please share them so we can check.

crivi30, 2025-05-29:
Here are the steps I've done so far:

1. I registered a delegate using the example implementation for the 1kW event trigger.

2. The attribute TestEventTriggersEnabled was successfully set to True.

3. To proceed with the next step in the test harness, I provided a custom enable_key in the PIXIT as an argument.

    I set PIXIT.EPM.TEST_EVENT_TRIGGER to 0091000000000001 and passed the key PIXIT.TEST_EVENT_TRIGGER_KEY:00112233445566778899aabbccddeeff.

    However, the test harness appears to ignore this key and instead uses the default one.

From the logs, it’s clear that regardless of the custom key I input in the PIXIT, the enable key used by the test harness is always:

"enableKey" : "hex:000102030405060708090A0B0C0D0E0F"

Infact, if I load this default key (000102030405060708090A0B0C0D0E0F) directly into my firmware as the enable key, the test passes correctly.

So I’m wondering — is it possible I’m using the wrong format for the custom argument key in the PIXIT?
Or I can just leave everything as a defult for the test harness.. 

Thanks

Jerry-ESP, 2025-06-04:
In the latest release/v1.4 branch, this file exist https://github.com/espressif/esp-matter/tree/release/v1.4/components/esp_matter , if you can't find it in your version. please update your sdk

crivi30, 2026-01-05:
yes, thanks

closed 2026-01-05 by crivi30