espressif/esp-matter #1424: PIC Generator issue

bilalahmaddev, 2025-05-23:
PIC Generator issue

[image]

**Environment**
- ESP-Matter Commit Id: v1.3
- ESP-IDF Commit Id: v5.3

- Host Machine OS: Ubuntu 22.04

Jerry-ESP, 2025-05-26:
This pics generate tool have some issue in v1.3-branch, you can use the latest version in the master branch. And please follow the instruction of the guide doc:https://github.com/project-chip/connectedhomeip/tree/master/src/tools/PICS-generator#setup

Jerry-ESP, 2025-06-03:
The latest pics tool version can use for all matter versions. include v1.3, you can use the latest pics tool to generate your pics file and then build your firmware on v1.3, they don't contradict each other.