espressif/esp-matter #1427: Matter 1.4 Electrical power measurement cluster Delegate Class

bilalahmaddev, 2025-05-25:
For matter 1.4 electrical power measurement cluster do we need to implement delegate ourself as these functions are defined as virtual in  electrical power measurement server delegate class or esp-matter handles them? 

```
class Delegate
{
public:
    virtual ~Delegate() = default;

    void SetEndpointId(EndpointId aEndpoint) { mEndpointId = aEndpoint; }

    virtual PowerModeEnum GetPowerMode()          = 0;
    virtual uint8_t GetNumberOfMeasurementTypes() = 0;

    /* These functions are called by the ReadAttribute handler to iterate through lists
     * The cluster server will call Start<Type>Read to allow the delegate to create a temporary
     * lock on the data.
     * The delegate is expected to not change these values once Start<Type>Read has been called
     * until the End<Type>Read() has been called (e.g. releasing a lock on the data)
     */
    virtual CHIP_ERROR StartAccuracyRead()                                                     = 0;
    virtual CHIP_ERROR GetAccuracyByIndex(uint8_t, Structs::MeasurementAccuracyStruct::Type &) = 0;
    virtual CHIP_ERROR EndAccuracyRead()                                                       = 0;

    virtual CHIP_ERROR StartRangesRead()                                                 = 0;
    virtual CHIP_ERROR GetRangeByIndex(uint8_t, Structs::MeasurementRangeStruct::Type &) = 0;
    virtual CHIP_ERROR EndRangesRead()                                                   = 0;

    virtual CHIP_ERROR StartHarmonicCurrentsRead()                                                     = 0;
    virtual CHIP_ERROR GetHarmonicCurrentsByIndex(uint8_t, Structs::HarmonicMeasurementStruct::Type &) = 0;
    virtual CHIP_ERROR EndHarmonicCurrentsRead()                                                       = 0;

    virtual CHIP_ERROR StartHarmonicPhasesRead()                                                     = 0;
    virtual CHIP_ERROR GetHarmonicPhasesByIndex(uint8_t, Structs::HarmonicMeasurementStruct::Type &) = 0;
    virtual CHIP_ERROR EndHarmonicPhasesRead()                                                       = 0;

    virtual DataModel::Nullable<int64_t> GetVoltage()         = 0;
    virtual DataModel::Nullable<int64_t> GetActiveCurrent()   = 0;
    virtual DataModel::Nullable<int64_t> GetReactiveCurrent() = 0;
    virtual DataModel::Nullable<int64_t> GetApparentCurrent() = 0;
    virtual DataModel::Nullable<int64_t> GetActivePower()     = 0;
    virtual DataModel::Nullable<int64_t> GetReactivePower()   = 0;
    virtual DataModel::Nullable<int64_t> GetApparentPower()   = 0;
    virtual DataModel::Nullable<int64_t> GetRMSVoltage()      = 0;
    virtual DataModel::Nullable<int64_t> GetRMSCurrent()      = 0;
    virtual DataModel::Nullable<int64_t> GetRMSPower()        = 0;
    virtual DataModel::Nullable<int64_t> GetFrequency()       = 0;
    virtual DataModel::Nullable<int64_t> GetPowerFactor()     = 0;
    virtual DataModel::Nullable<int64_t> GetNeutralCurrent()  = 0;

protected:
    EndpointId mEndpointId = 0;
};
```
Similarly for electrical energy, energy evse clusters.

Jerry-ESP, 2025-05-26:
Yes, you need to implement the delegate yourself, esp-matter doesn't do it in the sdk, please refer to the implementation  in this example: https://github.com/project-chip/connectedhomeip/blob/master/examples/energy-management-app/energy-management-common/energy-reporting/src/ElectricalPowerMeasurementDelegate.cpp,  and you can refer to this discussion: https://github.com/espressif/esp-matter/issues/1102#issuecomment-2408704757

lboue, 2025-06-06:
You need to add your delegates in `CMakeLists.txt` like in the refrigerator example:

https://github.com/espressif/esp-matter/blob/78d3def6258b118326bf9ca1e8cce1112536e74a/examples/refrigerator/main/CMakeLists.txt#L3

Example:
```
                   ${MATTER_SDK_PATH}/examples/energy-management-app/energy-management-common/energy-reporting/src/ElectricalPowerMeasurementDelegate.cpp
```

I'm interested in an example for the rest of the code because mine doesn't work properly.

olavt, 2025-06-10:
@lboue I copied the source files to my project, so I don't think the issue is related to CMakeLists.txt.

closed 2025-06-17 by bilalahmaddev

lboue, 2026-09-22:
https://github.com/espressif/esp-matter/tree/ff9f07ecb0bf01d7a2db5428149ef16bf601774c/examples/all_device_types_app/main/electrical_measurement