espressif/esp-matter #1430: ESP32C6 Matter WiFi low power commissioning

JordyDH, 2025-05-26:
Hi,

I'm currently integrating Matter into a product with strict power constraints.
Our available power budget is approximately 100–120 mW, supplied via a 10–12 V input rail buffered with a 12,000 µF bulk capacitor, allowing for short bursts above the nominal budget.

[image]

As shown in the PPK2 trace above, once Wi-Fi is connected, the ESP32-C6 no longer enters sleep mode. I've attempted to set the Wi-Fi power save mode to WIFI_PS_MAX_MODEM, but this doesn't appear to reduce the average power usage or allow the core to sleep.

My goal is to reduce idle power consumption during commissioning and after by enabling proper low-power states.

Questions:

Is there a recommended approach or known patch to slow down commissioning to allow the CPU and radio to enter sleep more aggressively?

Are there best practices or examples for using ESP-Matter with Wi-Fi in low-power scenarios?

Any help would be greatly appreciated.

JordyDH, 2025-05-27:
[image]

So with these patches the core goes more in sleep.
@DejinChen I still have a segment where the core doesn't go to sleep

[sdkconfig.copy.txt](https://github.com/user-attachments/files/20460123/sdkconfig.copy.txt)
Here is my sdkconfig if it could help.

DejinChen, 2025-05-28:
Generally, if no other tasks are running, it will run into idle task and go to light sleep, and any packets sending and receiving will leading the core active. So it won't affect the normal WiFi connecting and commissioning mode. As for the reason for this segment doesn't go to sleep, it may due to the frequently packets sending and receiving.

JordyDH, 2025-05-28:
Hi @DejinChen

Thanks for your previous response! It clarified a lot about idle and light sleep behavior during Wi-Fi activity.

I'm currently observing a segment where the core doesn’t enter sleep as expected—likely due to continuous packet activity. To reduce power draw, I’m considering introducing intentional delays during the Matter Wi-Fi commissioning phase. The goal is to allow the input capacitor to recharge between bursts of activity.

Questions:

Is it compliant with the Matter specification to insert short delays (e.g., a few hundred ms) in the commissioning workflow to give the system time to idle?

If so, where would be the appropriate place in the Matter stack or flow to insert these delays without breaking spec or impacting interoperability?

DejinChen, 2025-05-29:
Hi Jordy, after some testing, we found that the core stays active during the Wi-Fi connection to the AP (RX has to be active to receive the info from the AP). This explains the approximately 2.15-second segment above where it doesn’t enter sleep mode. As for the delay in the commissioning workflow, it may not help the Wi-Fi connection process and we would not recommend adding it. Based on your description, it seems there's a constraint on short-term high current consumption. Could you share more details on the acceptable duration and current threshold?

JordyDH, 2025-06-11:
So i have been busy trying things out to make the commussioning bellow a average consumption of 120mW.
[image]
I have modified the wifi scan method with a custom one that does does the scan in batches, this is needed to reduce the time the CPU is active.

[image]
Here is a piece of the DC/DC circuit we use for the product. We have a 120mF input buffer cap that we use so we can draw more then 120mW for a short time. The max voltage is around 10V and we can go up to 5V until the DC/DC is close to its low voltage lockout. So that gives us a charge of 600mC.

[image]
I was also able to insert a ~10s delay right before the wifi wants to connect. This is done to give the buffer cap time to build up a charge again.

The thing is I need to find a method to "slow" down or spread out the following task after wifi is connected so I don't discharge the cap to much because right after the wifi connect there isn't much left in it.

Also good to note is that is input power is limited to 120mW when you draw more it drops it voltage until the C6 crashes.

DejinChen, 2025-06-12:
After wifi connect, device will send connect response to commissioner, so here we may also delay the [connect response](https://github.com/project-chip/connectedhomeip/blob/a08bc0df452704aa2a7588c3e61f0f1b5291d79a/src/platform/ESP32/NetworkCommissioningDriver.cpp#L284) (the delay plus the connecting time should not exceed the wifi connect network timeout attribute). After receiving the response, commissioner will try to find the device on wifi and does the remaining process, and device will process the message immediately once received. So a simple way to delay the following task to delay the [message process](https://github.com/project-chip/connectedhomeip/blob/a08bc0df452704aa2a7588c3e61f0f1b5291d79a/src/transport/raw/UDP.cpp#L128), however, large delays are not recommended.

JordyDH, 2025-06-19:
So when doing some more test it keeps going in brownout when CASE session is created
`
I (57261) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:e6b3:23ff:fe87:6908
I (57272) app_main: Event: Unknown (0)
I (57275) matter_manager: Interface IP Address changed
I (57280) app_main: Event: Interface IP Changed
I (57284) chip[DIS]: Updating services using commissioning mode 0
I (57291) chip[DIS]: CHIP minimal mDNS started advertising.
I (57306) chip[DIS]: Advertise operational node DE8FEA0F292E58D3-00000000092339EC
I (57307) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: DE8FEA0F292E58D3-00000000092339EC.
I (57315) chip[DIS]: mDNS service published: _matter._tcp
I (57320) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=2960/11 cm=0 cp=0
I (57329) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 28937F3004CD7274.
I (57344) chip[DIS]: mDNS service published: _matterc._udp
I (57346) chip[DIS]: Updating services using commissioning mode 0
I (57351) chip[DIS]: CHIP minimal mDNS started advertising.
I (57367) chip[DIS]: Advertise operational node DE8FEA0F292E58D3-00000000092339EC
I (57368) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: DE8FEA0F292E58D3-00000000092339EC.
I (57377) chip[DIS]: mDNS service published: _matter._tcp
I (57380) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=2960/11 cm=0 cp=0
I (57390) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 28937F3004CD7274.
I (57406) chip[DIS]: mDNS service published: _matterc._udp
I (57406) chip[SVR]: Server initialization complete
I (57410) app_main: Event: Unknown (0)
I (57419) chip[IM]: No subscriptions to resume
I (57420) app_main: Event: Unknown (0)
I (57942) chip[EM]: >>> [E:22292r S:0 M:223803481] (U) Msg RX from 0:A735D5512AD29582 [0000] to 0000000000000000 --- Type 0000:30 (SecureChannel:CASE_Sigma1) (B:196)
I (57994) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x408296f8
I (57995) chip[EM]: <<< [E:22292r S:0 M:45158762 (Ack:223803481)] (U) Msg TX from 0000000000000000 to 0:A735D5512AD29582 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (58014) chip[SC]: Received Sigma1 msg
I (58023) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00000000092339EC
I (58721) chip[EM]: <<< [E:22292r S:0 M:45158763 (Ack:223803481)] (U) Msg TX from 0000000000000000 to 0:A735D5512AD29582 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:31 (SecureChannel:CASE_Sigma2) (B:529)
I (58731) chip[EM]: ??1 [E:22292r S:0 M:45158763] (U) Msg Retransmission to 0:0000000000000000 scheduled for 2312ms from now [State:Active II:2000 AI:2000 AT:2000]
I (58744) chip[SC]: Sent Sigma2 msg
I (59187) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (59589) chip[EM]: >>> [E:22292r S:0 M:223803482 (Ack:45158763)] (U) Msg RX from 0:A735D5512AD29582 [0000] to 0000000000000000 --- Type 0000:32 (SecureChannel:CASE_Sigma3) (B:380)
I (59641) chip[EM]: <<< [E:22292r S:0 M:45158764 (Ack:223803482)] (U) Msg TX from 0000000000000000 to 0:A735D5512AD29582 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (59654) chip[SC]: Received Sigma3 msg
`

Is it possible to change CASESession::OnMessageReceived in CASESession.cpp and add there a "delay"

JordyDH, 2025-06-23:
[image]

```
I (65277) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (65329) chip[EM]: <<1 [E:22625r S:0 M:258296699] (U) Msg Retransmission to 0:0000000000000000
I (65330) chip[EM]: ??2 [E:22625r S:0 M:258296699] (U) Msg Retransmission to 0:0000000000000000 scheduled for 2557ms from now [State:Active II:2000 AI:2000 AT:2000]
I (65514) chip[IN]: UDP: Adding delay before processing message
I (65565) chip[IN]: UDP: Adding delay before processing message
I (67931) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (68057) chip[EM]: <<2 [E:22625r S:0 M:258296699] (U) Msg Retransmission to 0:0000000000000000
I (68108) chip[EM]: ??3 [E:22625r S:0 M:258296699] (U) Msg Retransmission to 0:0000000000000000 scheduled for 4246ms from now [State:Idle II:2000 AI:2000 AT:2000]
I (68114) chip[EM]: >>> [E:22625r S:0 M:223803565 (Ack:258296699)] (U) Msg RX from 0:27B8C75B8EF92D68 [0000] to 0000000000000000 --- Type 0000:32 (SecureChannel:CASE_Sigma3) (B:380)
I (68129) chip[EM]: <<< [E:22625r S:0 M:258296701 (Ack:223803565)] (U) Msg TX from 0000000000000000 to 0:27B8C75B8EF92D68 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (68148) chip[SC]: Received Sigma3 msg
E (68155) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (69106) chip[IN]: UDP: Adding delay before processing message
I (69107) chip[IN]: UDP: Adding delay before processing message
I (69167) chip[EM]: <<< [E:22625r S:0 M:258296702 (Ack:223803565)] (U) Msg TX from 0000000000000000 to 0:27B8C75B8EF92D68 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:40 (SecureChannel:StatusReport) (B:34)
I (69178) chip[EM]: ??1 [E:22625r S:0 M:258296702] (U) Msg Retransmission to 0:0000000000000000 scheduled for 2463ms from now [State:Active II:2000 AI:2000 AT:2000]
I (69192) chip[SC]: SecureSession[0x40827cc8, LSID:58154]: State change 'kEstablishing' --> 'kActive'
I (69200) chip[IN]: CASE Session established to peer: <00000000031FAC64, 1>
I (69208) matter_manager: ✅ CASE session established over IP
I (69212) app_main: Event: Unknown (0)
I (69565) chip[IN]: UDP: Adding delay before processing message
I (69566) chip[IN]: UDP: Adding delay before processing message
I (70487) NimBLE: GATT procedure initiated: indicate; 
I (70487) NimBLE: att_handle=18

I (70545) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (70727) chip[EM]: >>> [E:22625r S:0 M:223803565 (Ack:258296699)] (U) Msg RX from 0:27B8C75B8EF92D68 [0000] to 0000000000000000 --- Type 0000:32 (SecureChannel:CASE_Sigma3) (B:380)
I (70733) chip[EM]: <<< [E:22625r S:0 M:258296703 (Ack:223803565)] (U) Msg TX from 0000000000000000 to 0:27B8C75B8EF92D68 [0000] [UDP:[FE80::8FB:EE1F:E5AC:C3A4%st1]:58349] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (70753) chip[EM]: >>> [E:22625r S:0 M:223803566 (Ack:258296699)] (U) Msg RX from 0:27B8C75B8EF92D68 [0000] to 0000000000000000 --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (71641) chip[EM]: <<1 [E:22625r S:0 M:258296702] (U) Msg Retransmission to 0:0000000000000000
I (71642) chip[EM]: ??2 [E:22625r S:0 M:258296702] (U) Msg Retransmission to 0:0000000000000000 scheduled for 2405ms from now [State:Active II:2000 AI:2000 AT:2000]

```

After Sigma3 is done the CPU is to long active.
Do you know what happens after Sigma3 and the CASE session is establised?

DejinChen, 2025-06-24:
Actually, establishing a CASE doesn't block the core for a long time. You can verify this by using the chip-tool to send a command to the device—CASE is established during command processing.

During the active period mentioned above, the RF is mostly off, yet the core remains active. I'm not sure whether there are any tasks that might block the core.

JordyDH, 2025-06-24:
I got it commissioned today on the 120mW PSU.
Only this is that sometimes the code is in a state that blocks the CPU from going into sleep.

[image]
This is the expected behaviour, right in ConnectivityManagerImpl::DriveStationState() i have placed a delay of 5s before a connection attempt is made.

[image]
but sometimes the CPU stays awake and draws to much and then the core crashes due to brown out.

I have tried to disable the wifi before the delay and this can still happen sometimes so I know it is not due to wifi driver locks.

so inside ConnectivityManagerImpl::DriveStationState I have added this sleep before the connect.
```
        if (mWiFiStationMode == kWiFiStationMode_Enabled && IsWiFiStationProvisioned())
        {
            // Initiate a connection to the AP if we haven't done so before, or if enough
            // time has passed since the last attempt.
            if (mLastStationConnectFailTime == System::Clock::kZero ||
                now >= mLastStationConnectFailTime + mWiFiStationReconnectInterval)
            {
                ChipLogProgress(DeviceLayer, "Attempting to connect WiFi station interface");
                
                // Only apply delay once during entire runtime
                if (!sWifiDelayDone)
                {
                    ChipLogProgress(DeviceLayer, "ONEDGE - WIFI CONNECT - delay %d ms - START", WIFI_CONNECT_DELAY);
                    vTaskDelay(pdMS_TO_TICKS(WIFI_CONNECT_DELAY));
                    ChipLogProgress(DeviceLayer, "ONEDGE - WIFI CONNECT - delay %d ms - END", WIFI_CONNECT_DELAY);
                    
                    // Set flag to true so we never do this delay again
                    sWifiDelayDone = true;
                }
                else
                {
                    ChipLogProgress(DeviceLayer, "Delay already done once, skipping");
                }

                esp_err_t err = esp_wifi_connect();
```

DejinChen, 2025-06-24:
You might consider using vTaskGetRunTimeStats or other FreeRTOS APIs to check whether the CPU enters the idle task during the active period mentioned above.

If it does, it's possible that some peripherals are still active, which could also prevent the CPU from sleeping.

JordyDH, 2025-06-25:
So I have implemented the vTaskGetRunTimeStats before and after the sleep.
```
                // Only apply delay once during entire runtime
                if (!sWifiDelayDone)
                {
                    ChipLogProgress(DeviceLayer, "ONEDGE - WIFI CONNECT - delay %d ms - START", WIFI_CONNECT_DELAY);
                    
                    // Print runtime stats before delay to analyze CPU usage
                    #if configGENERATE_RUN_TIME_STATS
                    {
                        char *pcWriteBuffer = (char*)malloc(2048);
                        if (pcWriteBuffer != NULL) {
                            ChipLogProgress(DeviceLayer, "=== Task Runtime Stats Before WiFi Delay ===");
                            vTaskGetRunTimeStats(pcWriteBuffer);
                            // Print line by line to avoid log buffer overflow
                            char *line = strtok(pcWriteBuffer, "\n");
                            while (line != NULL) {
                                ChipLogProgress(DeviceLayer, "%s", line);
                                line = strtok(NULL, "\n");
                            }
                            ChipLogProgress(DeviceLayer, "=============================================");
                            free(pcWriteBuffer);
                        }
                    }
                    #else
                    ChipLogProgress(DeviceLayer, "Runtime stats not enabled. Enable configGENERATE_RUN_TIME_STATS in sdkconfig");
                    #endif
                    
                    // Print power management configuration
                    #if CONFIG_PM_ENABLE
                    {
                        esp_pm_config_t pm_config;
                        esp_err_t pm_err = esp_pm_get_configuration(&pm_config);
                        if (pm_err == ESP_OK) {
                            ChipLogProgress(DeviceLayer, "Power Management Config: max_freq=%d MHz, min_freq=%d MHz, light_sleep=%s",
                                          pm_config.max_freq_mhz, pm_config.min_freq_mhz, 
                                          pm_config.light_sleep_enable ? "enabled" : "disabled");
                        }
                    }
                    #else
                    ChipLogProgress(DeviceLayer, "Power Management is disabled in sdkconfig");
                    #endif
                    
                    // Check what's preventing sleep
                    ChipLogProgress(DeviceLayer, "Checking sleep blockers...");
                    esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
                    
                    vTaskDelay(pdMS_TO_TICKS(WIFI_CONNECT_DELAY));
                    
                    // Print runtime stats after delay to see what consumed CPU time
                    #if configGENERATE_RUN_TIME_STATS
                    {
                        char *pcWriteBuffer = (char*)malloc(2048);
                        if (pcWriteBuffer != NULL) {
                            ChipLogProgress(DeviceLayer, "=== Task Runtime Stats After WiFi Delay ===");
                            vTaskGetRunTimeStats(pcWriteBuffer);
                            // Print line by line to avoid log buffer overflow
                            char *line = strtok(pcWriteBuffer, "\n");
                            while (line != NULL) {
                                ChipLogProgress(DeviceLayer, "%s", line);
                                line = strtok(NULL, "\n");
                            }
                            ChipLogProgress(DeviceLayer, "============================================");
                            free(pcWriteBuffer);
                        }
                    }
                    #endif
                    
                    ChipLogProgress(DeviceLayer, "ONEDGE - WIFI CONNECT - delay %d ms - END", WIFI_CONNECT_DELAY);

```

and this is the result in the logs:

```
I (47835) chip[DL]: Attempting to connect WiFi station interface
I (47839) chip[DL]: ONEDGE - WIFI CONNECT - delay 5000 ms - START
I (47845) chip[DL]: === Task Runtime Stats Before WiFi Delay ===
I (47853) chip[DL]: CHIP           	6173997		13%
I (47855) chip[DL]: IDLE           	35180578		76%
I (47859) chip[DL]: main           	613945		1%
I (47863) chip[DL]: Tmr Svc        	21		<1%
I (47867) chip[DL]: tiT            	11569		<1%
I (47872) chip[DL]: nimble_host    	2040017		4%
I (47876) chip[DL]: esp_timer      	96908		<1%
I (47880) chip[DL]: wifi           	342195		<1%
I (47885) chip[DL]: ble_ll_task    	1251110		2%
I (47889) chip[DL]: sys_evt        	4014		<1%
I (47893) chip[DL]: =============================================
I (47899) chip[DL]: Power Management Config: max_freq=80 MHz, min_freq=40 MHz, light_sleep=enabled
I (47908) chip[DL]: Checking sleep blockers...
I (52912) chip[DL]: === Task Runtime Stats After WiFi Delay ===
I (52915) chip[DL]: CHIP           	6248076		12%
I (52915) chip[DL]: IDLE           	40160664		78%
I (52916) chip[DL]: main           	613961		1%
I (52920) chip[DL]: Tmr Svc        	21		<1%
I (52924) chip[DL]: tiT            	12073		<1%
I (52928) chip[DL]: wifi           	342195		<1%
I (52932) chip[DL]: esp_timer      	97190		<1%
I (52936) chip[DL]: ble_ll_task    	1270240		2%
I (52941) chip[DL]: sys_evt        	4014		<1%
I (52945) chip[DL]: nimble_host    	2040017		4%
I (52949) chip[DL]: ============================================
I (52955) chip[DL]: ONEDGE - WIFI CONNECT - delay 5000 ms - END
I (52962) chip[DL]: WiFi station state change: NotConnected -> Connecting
W (52967) wifi:Haven't to connect to a suitable AP now!

```

DejinChen, 2025-06-25:
The result of vTaskGetRunTimeStats shows that the CPU spends most of its time on the IDLE task. If no peripherals are holding the pm_lock, the CPU should enter sleep during this state. Therefore, might check whether the CPU truly enters the IDLE task, or if it remains in the IDLE task while some peripherals are preventing sleep by holding the pm_lock during the active period mentioned above.

JordyDH, 2025-06-26:
@DejinChen thanks for all the help, rigth now i have a build that works 90% of the time.
Sometimes the core stays awake when i delay the wifi connect but this happens a lot less, i changed it from vTaskDelay to the inbuild timer schudeler of CHIP.

What I have done is:

- Slow down UDP
- Slow down BLE
- Slow down CASE
- Slow down PASE
- Added a segmented wifi scan
- Added pm settings in the places wifi is setup

[image]
Right now I have the problem when the AP is not available anymore or the wifi connection drops that the core stays awake and discharges my caps.

Is there a way or method to disable wifi and then retry the connect later without having the core awake like this?

DejinChen, 2025-06-26:
Setting `SetWiFiStationMode(kWiFiStationMode_ApplicationControlled)` disables the automatic Wi-Fi reconnection process. To re-enable it, use `SetWiFiStationMode(kWiFiStationMode_Enabled)`.
Additionally, the Wi-Fi reconnection interval (CONFIG_WIFI_STATION_RECONNECT_INTERVAL) can be adjusted to suit your application's requirements.

JordyDH, 2025-07-03:
> I got it commissioned today on the 120mW PSU. Only this is that sometimes the code is in a state that blocks the CPU from going into sleep.
>
> [image] This is the expected behaviour, right in [...]

So I have removed the vTaskDelay from here and used the CHIP timer scheduler. But is still happens that the core is busy.
Is it possible that I can share my changes fro you to have a look?

JordyDH, 2025-07-09:
Is it possible that some of the problems we have can be solved by switching to IDF version 5.4.2?

DejinChen, 2025-07-10:
Switching to IDF 5.4.2 may not resolve the inactive time issue, as disconnecting Wi-Fi may reset the inactive time. Currently, the  workaround is to set the inactive time before each call to esp_wifi_connect.
For the issue where the CPU does not enter sleep, we need to investigate the root cause, or you may provide steps to reproduce it.
In any case, switching to IDF 5.4.2 is acceptable.