espressif/esp-matter #1432: light example, The legacy RMT driver is deprecated?

olavt, 2025-05-26:
When building the light example, you get:

-- Build files have been written to: /Users/olavt/Source/Repos/MatterDevices/ESP32C6/air-quality/build/bootloader
[1375/1643] Building C object esp-idf/espressif__led_strip/CMakeFiles/__idf_espressif__led_strip.dir/src/led_strip_rmt_ws2812.c.obj
In file included from /Users/olavt/Source/Repos/MatterDevices/ESP32C6/air-quality/managed_components/espressif__led_strip/src/led_strip_rmt_ws2812.c:20:
/Users/olavt/esp/v5.4.1/esp-idf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
[1409/1643] Building C object esp-idf/led_driver/CMakeFiles/__idf_led_driver.dir/ws2812/led_driver.c.obj
In file included from /Users/olavt/.espressif/esp-matter/device_hal/led_driver/ws2812/led_driver.c:15:
/Users/olavt/esp/v5.4.1/esp-idf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
[

Why is the example using a deprecated led strip driver?

Shouldn't it be updated?

olavt, 2025-05-28:
What are the steps to upgrade to a newer version of the led strip driver in VS Code?

Jerry-ESP, 2025-05-30:
No, consider the firmware's RAM size and flash size, we don't update the driver version automatically