espressif/esp-matter #1433: InstrFetchProhibited Error During Matter TC-DA-1.5 (IDFGH-15375)

emesamwong, 2025-05-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

During the Matter Certification Test Plan TC-DA-1.5, the DUT was successfully commissioned for the first step. 
And for the second fabric pairing, DUT crashed with the InstrFetchProhibited Error. 

So far, I have tried to execute TC-DA-1.5 Testing four times and crashed once. 
I think the root cause is not the test case since not trigger the DUT to crash every time.
How can I locate the ROOT cause of the crash? 
Could I get any more information to locate the issue for the Error message? 
[image]

TC-DA-1.5 Failure Log: 
[TC-DA-1.5_Failure_DUT_reboot.log](https://github.com/user-attachments/files/20447728/TC-DA-1.5_Failure_DUT_reboot.log)

DUT Crash Log:
[TC-DA-1.5 Failure DUT Log.log](https://github.com/user-attachments/files/20447734/TC-DA-1.5.Failure.DUT.Log.log)

TC-DA-1.5 Success Log:
[TC-DA-1.5 PASS.log](https://github.com/user-attachments/files/20447741/TC-DA-1.5.PASS.log)

DUT Normal Log:
[DUT Normal Log.log](https://github.com/user-attachments/files/20447800/DUT.Normal.Log.log)

Please let me know if you need more information

bilalahmaddev, 2025-05-27:
I executed this test many times for different devices and never seen this crash with  TC-DA-1.5 with either 1.3 or 1.4.

dhrishi, 2025-06-06:
@emesamwong Is there an update that you can share? Please close the issue if it works now

closed 2025-09-13 by dhrishi