espressif/esp-matter #1435: Adding Optional Attributes (Voltage, ActiveCurrent) to Electrical Power Measurement Cluster

crivi30, 2025-05-29:
Hi everyone,

I'm trying to manually add optional attributes to the *Electrical Power Measurement* cluster (0x0000_0090) in my Matter device firmware.

The cluster works fine for the mandatory `ActivePower` attribute , and it's correctly shown when I read it using `chip-tool`. However, when I try to access the optional attributes `Voltage`  and `ActiveCurrent` , I get an `UNSUPPORTED_ATTRIBUTE` error, even though:

- There are no errors or warnings in the firmware logs in the attribute creation;
- The `AttributeList` returned by `chip-tool` **does** include both optional attributes (Voltage and ActiveCurrent), as shown below:

```
[1748515313.173] [243997:244028] [TOO] Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_FFFB DataVersion: 1980714119
[1748515313.173] [243997:244028] [TOO]   AttributeList: 11 entries
[1748515313.173] [243997:244028] [TOO]     [1]: 65528 (GeneratedCommandList)
[1748515313.173] [243997:244028] [TOO]     [2]: 65529 (AcceptedCommandList)
[1748515313.173] [243997:244028] [TOO]     [3]: 65531 (AttributeList)
[1748515313.173] [243997:244028] [TOO]     [4]: 65532 (FeatureMap)
[1748515313.173] [243997:244028] [TOO]     [5]: 0 (PowerMode)
[1748515313.173] [243997:244028] [TOO]     [6]: 1 (NumberOfMeasurementTypes)
[1748515313.173] [243997:244028] [TOO]     [7]: 2 (Accuracy)
[1748515313.173] [243997:244028] [TOO]     [8]: 8 (ActivePower)
[1748515313.173] [243997:244028] [TOO]     [9]: 65533 (ClusterRevision)
[1748515313.173] [243997:244028] [TOO]     [10]: 4 (Voltage)
[1748515313.173] [243997:244028] [TOO]     [11]: 5 (ActiveCurrent)
```

But when I try to read them, I get this error:

`Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)`

I have Used the following code to add the attributes:

```
app_electrical_sensor->endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, NULL);

// Add optional attribute to the ElectricalPowerMeasurement cluster
cluster_t *elec_power_measurement_cluster = cluster::get(app_electrical_sensor->endpoint, ElectricalPowerMeasurement::Id); 

nullable<int64_t> voltage = 0, active_current = 0;
electrical_power_measurement::attribute::create_voltage(elec_power_measurement_cluster, voltage);
electrical_power_measurement::attribute::create_active_current(elec_power_measurement_cluster, active_current);

```

**Question:**
What am I missing in the implementation? How can I properly add and support these two optional attributes (Voltage, ActiveCurrent) so they are readable by chip-tool?

Thanks in advance for any help or insight!

pimpalemahesh, 2025-06-02:
Hi @crivi30,

I tested the changes you mentioned and was able to successfully read both the Voltage and ActiveCurrent attributes using chip-tool—no issues on my end.

Please ensure you're using the correct command to read the attributes:
`chip-tool electricalpowermeasurement read voltage 1 1`
`chip-tool electricalpowermeasurement read active-current 1 1`

In my setup, both the node ID and endpoint ID are 1. Make sure you’re using the appropriate values for your setup.
If the issue still persists, could you please share the following details to help with further debugging:

- esp-matter commit ID
- esp-idf commit ID
- Your sdkconfig
- Relevant chip-tool and device-side logs

crivi30, 2025-06-03:
Hi @pimpalemahesh,

I confirm that I’m using the correct chip-tool command to read.
For building the firmware, I’m using a recent Docker image — specifically espressif/esp-matter:release-v1.4.

To ensure reproducibility, I’ve attached the docker image inspect output, which includes the full image digest and build metadata.

Let me know if you need more details.

Best regards

[docker_image_inspect.txt](https://github.com/user-attachments/files/20564683/docker_image_inspect.txt)

closed 2025-06-06 by crivi30