espressif/esp-matter #1438: OTA Update requestor cluster Info

bilalahmaddev, 2025-05-30:
Where can I fine information related to OTA update requestor cluster in Matter specification? I do not find any information in device cluster library or core specification? 

Also there is no XML file for this cluster on reference XML available in CSA members account. From where to get this? 

Can you please share if I am missing something?

pimpalemahesh, 2025-06-02:
@bilalahmaddev The OTA Software Update Requestor Cluster is documented in the Matter Core Specification. You can find the relevant details in Section `11.20.7.  OTA Software Update Requestor Cluster` of the Matter 1.4 Core Specification, available [here](https://csa-iot.org/wp-content/uploads/2024/11/24-27349-006_Matter-1.4-Core-Specification.pdf).

As for the XML reference:

The spec-generated XML for the OTA Requestor Cluster can be found in the Connectedhomeip repository [here](https://github.com/project-chip/connectedhomeip/blob/master/data_model/master/clusters/OTARequestor.xml).
Additionally, you can refer to the handwritten XML definition [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/chip-ota.xml).

bilalahmaddev, 2025-06-02:
@pimpalemahesh Thanks but these XML do not have any PICS codes. When you upload it on PICS tool it show empty.

VaishaliAvhale, 2025-06-02:
Yes, I've had a look, and it's not present in the Matter v1.3 PICS XML folder. I will check with CSA and update you accordingly.

bilalahmaddev, 2025-06-03:
I do not have access to chip test plans repo. But I think they already know about this because PIC verification excel file already have this issue mentioned.

closed 2025-06-06 by dhrishi