espressif/esp-matter #1440: Thermostat temperature step

dzungpv, 2025-06-01:
With HomeKit I can easy add temperature step when change temperature like 0.5 or 1.0, compatible with Air conditioner. But I can not find any way to set this in esp matter.
Default step when add to Apple Home app is 0.5.

I try this code: 
```
    // Temperature step
    if (_temp_step_half) { // 0.5 deegree
        esp_matter::cluster::thermostat::attribute::create_setpoint_change_amount(thermostat_cluster, 0.5*100);
    } else { // 1.0 deegree
        esp_matter::cluster::thermostat::attribute::create_setpoint_change_amount(thermostat_cluster, 1*100);
    }
    // Temperature step source
    esp_matter::cluster::thermostat::attribute::create_setpoint_change_source(thermostat_cluster, 0);  // 0 manual

```

olavt, 2025-06-01:
What did you expect should happen based on setting these attributes and what did you observe happened?

dzungpv, 2025-06-02:
> What did you expect should happen based on setting these attributes and what did you observe happened?

I mean when changing temperature, it change each time 0.5 or 1 degree base on the Air conditioner support, but Apple Home app only change 0.5 each time

olavt, 2025-06-02:
I don’t think these attributtes have that purpose.

dzungpv, 2025-06-02:
> I don’t think these attributtes have that purpose.

Yes, I am wrong, it is fixed by Home app, It is available with Home Assistant app only

closed 2025-06-02 by dzungpv