espressif/esp-matter #1443: Test Event Trigger Telated test cases for electrical power and energy measurement, EVSE energy

bilalahmaddev, 2025-06-03:
We’re using the Matter SDK and need to assign TestEventTriggerDelegate instances per endpoint to handle cluster-specific test events for electrical power and energy measurement, EVSE energy. What's the recommended approach to associate delegates with each endpoint separately?

Secondly, I do not see [/esp_matter_test_event_trigger.cpp](https://github.com/espressif/esp-matter/blob/057ac6a3e06f29a8a6f0e942228d01064e6942d3/components/esp_matter/esp_matter_test_event_trigger.cpp#L23) file while matter v1.4 as a component in esp-idf?

Jerry-ESP, 2025-06-05:
1. You can refer to this implementation: https://github.com/project-chip/connectedhomeip/blob/b1fc7197d197c6721a93b7d59efb5d2305ec6735/examples/platform/esp32/common/Esp32AppServer.cpp#L153-L193,  define your delegate use SimpleTestEventTriggerDelegate class, init enable key,  call AddHandler to add your test event implementation. 
Then call the esp_matter::test_event_trigger::set_delegate to set the delegate. 

For your current  code,  the HandleTestEventTrigger function will not be called, you can define your TestEventTriggerHandler like this :  https://github.com/project-chip/connectedhomeip/blob/b1fc7197d197c6721a93b7d59efb5d2305ec6735/src/app/clusters/energy-evse-server/EnergyEvseTestEventTriggerHandler.h#L74-L92

2.  To set the enable key, you can refer to this code: https://github.com/project-chip/connectedhomeip/blob/b1fc7197d197c6721a93b7d59efb5d2305ec6735/examples/platform/esp32/common/Esp32AppServer.cpp#L153-L161

bilalahmaddev, 2025-06-10:
@Jerry-ESP Thanks I am able to implement by following above links but our device has multiple endpoints for electrical power and electrical energy measurement clusters. How to use test event trigger separately for each endpoint, like how to register handlers for each endpoint?

Jerry-ESP, 2025-06-11:
I think you can define different events for different endpoints