espressif/esp-matter #1444: Electrical Sensor examples for ESP-Matter?

olavt, 2025-06-03:
Are there any examples on how to implement Electrical Sensor on ESP-Matter using code (no ZAP)?

I find it a bit hard to understand how to structure the code using the ElectricalPowerMeasurement::Delegate classes.

jadhavrohit924, 2025-06-04:
We are working on it.

olavt, 2025-06-04:
Could you outline an overview of the required steps?

jadhavrohit924, 2025-06-25:
> 1) Why aren't all the mandatory for the device type Electrical Sensor created by esp_matter::endpoint::electrical_sensor::create?

Esp-matter component by default creates mandatory bits as per spec only.

> 2) Where is the chip::app::Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurement::Delegate wired up (if it is)?

PTAL https://github.com/espressif/esp-matter/blob/main/examples/all_device_types_app/main/electrical_measurement/electrical_measurement.cpp#L183-L221

> 3) Where is the Accuracy attribute of the Electrical Energy Measurement Cluster set / initialized?

PTAL https://github.com/espressif/esp-matter/blob/main/examples/all_device_types_app/main/electrical_measurement/electrical_measurement.cpp#L158-L162

jadhavrohit924, 2025-06-25:
`O.a+` conformance means that at least one of them is mandatory. So `A` OR `B` OR `A, B` are possible. It's on application to decide.

jadhavrohit924, 2025-06-30:
We have updated the logic of creating features, features are now configurable in the application, please pull the latest main.

olavt, 2025-06-30:
@jadhavrohit924 Do you have any details on how this is configured? I would actually prefer to have the ability to set the features in code.