espressif/esp-matter #1447: Issue Adding Optional Attributes to PowerMeasurement Cluster When Delegate Is Set

crivi30, 2025-06-06:
Hi,

I've encountered an issue when trying to add optional attributes to the ElectricalPowerMeasurement cluster after registering a delegate. The problem appears when I register a delegate and then attempt to add attributes like voltage and current. Here's a simplified version of the setup that triggers the issue:

```
// Add Power Measurement Delegate
app_electrical_sensor->pm_delegate = new Clusters::ElectricalPowerMeasurement::ElectricalPowerMeasurementDelegate();
app_electrical_sensor->pm_delegate->SetActivePower(DataModel::Nullable<int64_t>(0));
app_electrical_sensor->pm_delegate->SetPowerMode(ElectricalPowerMeasurement::PowerModeEnum::kAc);
electrical_sensor_config.electrical_power_measurement.delegate = app_electrical_sensor->pm_delegate;

app_electrical_sensor->endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, NULL);

// Optional attributes for ElectricalPowerMeasurement
cluster_t *elec_power_measurement_cluster = cluster::get(app_electrical_sensor->endpoint, ElectricalPowerMeasurement::Id);
nullable<int64_t> voltage = 0, active_current = 0;
electrical_power_measurement::attribute::create_voltage(elec_power_measurement_cluster, voltage);
app_electrical_sensor->pm_delegate->SetVoltage(DataModel::Nullable<int64_t>(0));
electrical_power_measurement::attribute::create_active_current(elec_power_measurement_cluster, active_current);
app_electrical_sensor->pm_delegate->SetActiveCurrent(DataModel::Nullable<int64_t>(0));
```

Once the delegate is registered, the optional attributes voltage and active current don't seem to be handled correctly. I'm unsure whether this is an issue with the attribute creation, the delegate implementation, or the timing/order of operations.

I also tried to create in Instance class of if, without any luck:

```
app_electrical_sensor->epmInstance = new ElectricalPowerMeasurement::ElectricalPowerMeasurementInstance(
        chip::EndpointId(app_electrical_sensor->endpoint_id),
        *app_electrical_sensor->pm_delegate,
        chip::BitMask<ElectricalPowerMeasurement::Feature, uint32_t>(ElectricalPowerMeasurement::Feature::kDirectCurrent, ElectricalPowerMeasurement::Feature::kAlternatingCurrent),
        chip::BitMask<ElectricalPowerMeasurement::OptionalAttributes, uint32_t>(
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeVoltage,
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeActiveCurrent, ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeReactiveCurrent,
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeApparentCurrent, ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeReactivePower,
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeApparentPower, ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSVoltage,
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSCurrent, ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSPower,
            ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributePowerFactor, ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeNeutralCurrent));

app_electrical_sensor->epmInstance->Init();
```

Has anyone else encountered this behavior? Is there a recommended approach to dynamically adding optional attributes when using a delegate with the PowerMeasurement cluster?

Thanks in advance for your help!

jadhavrohit924, 2025-06-06:
> So, Is there a way to dynamically add optional attributes to the ElectricalPowerMeasurement::Instance after/before it's initialized? or what’s the recommended pattern for toggling optional attributes?

No, however you can create your own instance of the cluster to add optional attributes/features instead of setting the delegate.

I mean you can skip this line, and create cluster instance on your own.
`electrical_sensor_config.electrical_power_measurement.delegate = app_electrical_sensor->pm_delegate;`

crivi30, 2025-06-09:
Closing this to open a more targeted issue for the specific problem I'm facing. Thanks!