espressif/esp-matter #1449: I am trying sensor example for Esp32-c3 but got this error, how to fix it

hoanbk65, 2025-06-07:
I (27) boot: ESP-IDF v5.5-dev-3236-g465b159cd8 2nd stage bootloader
I (27) boot: compile time Jun 5 2025 10:03:21
I (28) boot: chip revision: v0.0
I (29) boot: efuse block revision: v0.1
I (32) boot.esp32c6: SPI Speed : 80MHz
I (36) boot.esp32c6: SPI Mode : DIO
I (40) boot.esp32c6: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (50) boot: ## Label Usage Type ST Offset Length
I (57) boot: 0 esp_secure_cert unknown 3f 06 0000d000 00002000
I (63) boot: 1 nvs WiFi data 01 02 00010000 0000c000
I (70) boot: 2 nvs_keys NVS keys 01 04 0001c000 00001000
I (76) boot: 3 otadata OTA data 01 00 0001d000 00002000
I (83) boot: 4 phy_init RF data 01 01 0001f000 00001000
I (89) boot: 5 ota_0 OTA app 00 10 00020000 001e0000
I (96) boot: 6 ota_1 OTA app 00 11 00200000 001e0000
I (102) boot: 7 fctry WiFi data 01 02 003e0000 00006000
I (109) boot: End of partition table
I (113) esp_image: segment 0: paddr=00020020 vaddr=42140020 size=3a180h (237952) map
I (205) esp_image: segment 1: paddr=0005a1a8 vaddr=40800000 size=05e70h ( 24176) load
I (216) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=13c808h (1296392) map
I (679) esp_image: segment 3: paddr=0019c830 vaddr=40805e70 size=1a018h (106520) load
I (724) esp_image: segment 4: paddr=001b6850 vaddr=4081fe90 size=03d2ch ( 15660) load
I (746) boot: Loaded app from partition at offset 0x20000
I (746) boot: Disabling RNG early entropy source...
I (757) cpu_start: Unicore app
I (766) cpu_start: Pro cpu start user code
I (766) cpu_start: cpu freq: 160000000 Hz
I (766) app_init: Application information:
I (766) app_init: Project name: sensors
I (770) app_init: App version: 1.0
I (773) app_init: Compile time: Jun 5 2025 10:03:05
I (778) app_init: ELF file SHA256: 031ad6f1d...
I (782) app_init: ESP-IDF: v5.5-dev-3236-g465b159cd8
I (788) efuse_init: Min chip rev: v0.0
I (792) efuse_init: Max chip rev: v0.99
I (796) efuse_init: Chip rev: v0.0
I (800) heap_init: Initializing. RAM available for dynamic allocation:
I (806) heap_init: At 408381E0 len 00044430 (273 KiB): RAM
I (811) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (816) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
W (822) spi_flash: GigaDevice detected but related driver is not linked, please check option SPI_FLASH_SUPPORT_GD_CHIP
I (832) spi_flash: detected chip: generic
I (836) spi_flash: flash io: dio
E (839) i2c: CONFLICT! driver_ng is not allowed to be used with this old driverE (839) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x420e2f1d on core 0
--- 0x420e2f1d: check_i2c_driver_conflict at /home/hoandz/esp/esp-idf/components/driver/i2c/i2c.c:1738

--- Stack dump detected
Core 0 register dump:
MEPC : 0x40800804 RA : 0x4080d4b8 SP : 0x4087e160 GP : 0x40820694
--- 0x40800804: panic_abort at /home/hoandz/esp/esp-idf/components/esp_system/panic.c:482
--- 0x4080d4b8: __ubsan_include at /home/hoandz/esp/esp-idf/components/esp_system/ubsan.c:311

TP : 0x00000000 T0 : 0x37363534 T1 : 0x7271706f T2 : 0x33323130
S0/FP : 0x4087e19c S1 : 0x4087e19c A0 : 0x4087e19c A1 : 0x4087e17e
A2 : 0x00000000 A3 : 0x4087e1c9 A4 : 0x00000001 A5 : 0x40836000
A6 : 0x00000000 A7 : 0x76757473 S2 : 0x4087e180 S3 : 0x408002da
--- 0x408002da: call_start_cpu0 at /home/hoandz/esp/esp-idf/components/esp_system/port/cpu_start.c:357

S4 : 0x42000000 S5 : 0x43000000 S6 : 0x00060020 S7 : 0x0013c808
S8 : 0x0003a180 S9 : 0x42140000 S10 : 0x00000000 S11 : 0x00000000
T3 : 0x6e6d6c6b T4 : 0x6a696867 T5 : 0x66656463 T6 : 0x62613938
MSTATUS : 0x00001881 MTVEC : 0x40800001 MCAUSE : 0x00000002 MTVAL : 0x00000000
--- 0x40800001: _vector_table at /home/hoandz/esp/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000

--- Backtrace:

add symbol table from file "/home/hoandz/doan/sensors/build/bootloader/bootloader.elf"
panic_abort (details=details@entry=0x4087e19c "abort() was called at PC 0x420e2f1d on core 0") at /home/hoandz/esp/esp-idf/components/esp_system/panic.c:482
482 asm("unimp"); // should be an invalid operation on RISC-V targets
#0 panic_abort (details=details@entry=0x4087e19c "abort() was called at PC 0x420e2f1d on core 0") at /home/hoandz/esp/esp-idf/components/esp_system/panic.c:482
https://github.com/espressif/esp-matter/issues/1 0x4080d4b8 in esp_system_abort (details=details@entry=0x4087e19c "abort() was called at PC 0x420e2f1d on core 0") at /home/hoandz/esp/esp-idf/components/esp_system/port/esp_system_chip.c:87
https://github.com/espressif/esp-matter/issues/2 0x40814580 in abort () at /home/hoandz/esp/esp-idf/components/newlib/src/abort.c:38
https://github.com/espressif/esp-matter/issues/3 0x420e2f20 in check_i2c_driver_conflict () at /home/hoandz/esp/esp-idf/components/driver/i2c/i2c.c:1738
https://github.com/espressif/esp-matter/issues/4 0x420013cc in do_global_ctors () at /home/hoandz/esp/esp-idf/components/esp_system/startup.c:104
https://github.com/espressif/esp-matter/issues/5 start_cpu0_default () at /home/hoandz/esp/esp-idf/components/esp_system/startup.c:206
https://github.com/espressif/esp-matter/issues/6 0x4086fb6e in set_cache_and_start_app (drom_addr=, drom_load_addr=, drom_size=237984, irom_addr=393248, irom_load_addr=, irom_size=1296424, data=) at /home/hoandz/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:1158
https://github.com/espressif/esp-matter/issues/7 unpack_load_app (data=0x4087e138) at /home/hoandz/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:848
https://github.com/espressif/esp-matter/issues/8 load_image (image_data=image_data@entry=0x4087e268) at /home/hoandz/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:803
https://github.com/espressif/esp-matter/issues/9 0x4087003c in bootloader_utility_load_boot_image (bs=bs@entry=0x4087e3a0, start_index=start_index@entry=0) at /home/hoandz/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:605
https://github.com/espressif/esp-matter/issues/10 0x4086b99c in call_start_cpu0 () at /home/hoandz/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:62
https://github.com/espressif/esp-matter/issues/11 0x4001b1c0 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

ELF file SHA256: 031ad6f1d

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x6c (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROMSPIWP:0xee
mode:DIO, clock div:2
load:0x40875730,len:0x1744
load:0x4086b910,len:0xe78
--- 0x4086b910: esp_bootloader_get_description at /home/hoandz/esp/esp-idf/components/esp_bootloader_format/esp_bootloader_desc.c:39 (discriminator 1)

load:0x4086e610,len:0x31a8
--- 0x4086e610: esp_flash_encryption_enabled at /home/hoandz/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c:89

entry 0x4086b91a
--- 0x4086b91a: call_start_cpu0 at /home/hoandz/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

pimpalemahesh, 2025-06-11:
@hoanbk65 Thank you for reporting the issue. There is indeed a crash occurring in the sensors example. We will investigate and provide a fix soon.

hoanbk65, 2025-06-14:
@pimpalemahesh can you give me the fix in esp-idf 5.4 version

pimpalemahesh, 2025-06-15:
@hoanbk65 You can apply this patch to the ESP-IDF repository to skip the I2S legacy conflict check:
[i2s_conflict_skip.txt](https://github.com/user-attachments/files/20743741/i2s_conflict_skip.txt)
```

cd /path/to/esp-idf
git apply i2s_conflict_skip.txt
source /path/to/esp-idf/export.sh

```

pimpalemahesh, 2025-06-17:
@hoanbk65 Have you had a chance to try the patch? Please ensure that `CONFIG_I2C_SKIP_LEGACY_CONFLICT_CHECK=y` is enabled, as I mentioned earlier

hoanbk65, 2025-06-17:
ok my friend thanks

hoanbk65, 2025-06-17:
i tried it and it worked thank you

pimpalemahesh, 2025-06-17:
@hoanbk65 Please close this issue!

closed 2025-06-17 by hoanbk65