espressif/esp-matter #1451: How to create a custom instance of ElectricalPowerMeasurement with optional attributes?

crivi30, 2025-06-09:
I'm trying to create a custom instance of the ElectricalPowerMeasurement Cluster to enable support for optional attributes, as mentioned in issue #1447 .

According to that issue, the only way to add optional attributes is to set up my own Cluster instance. I've attempted to do that with the following code, but I'm encountering an error when trying to read the accuracy attribute:

`Response Failure: IM Error 0x00000501: General error: 0x01 (FAILURE)`

Here is the code I'm using to create and initialize the instance:

```
// Create the instance
app_electrical_sensor->epmInstance = new ElectricalPowerMeasurement::ElectricalPowerMeasurementInstance(
    chip::EndpointId(app_electrical_sensor->endpoint_id),
    *app_electrical_sensor->pm_delegate,
    chip::BitMask<ElectricalPowerMeasurement::Feature, uint32_t>(
        ElectricalPowerMeasurement::Feature::kDirectCurrent,
        ElectricalPowerMeasurement::Feature::kAlternatingCurrent),
    chip::BitMask<ElectricalPowerMeasurement::OptionalAttributes, uint32_t>(
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeVoltage,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeActiveCurrent,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeReactiveCurrent,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeApparentCurrent,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeReactivePower,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeApparentPower,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSVoltage,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSCurrent,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeRMSPower,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributePowerFactor,
        ElectricalPowerMeasurement::OptionalAttributes::kOptionalAttributeNeutralCurrent));

app_electrical_sensor->epmInstance->Init();

```
Am I missing any additional steps to properly initialize or register the instance?
How should I correctly configure it so the accuracy is available and can be read successfully?

Thanks in advance for your support!

jadhavrohit924, 2025-06-11:
@crivi30 We have created an example for Electrical Power and Energy Measurement cluster. It is in internal review and will be out soon.

crivi30, 2025-06-25:
Hi,

Thanks so much for the quick response and the helpful example!

When I bring the code into our project, which is on Matter v1.4, it no longer works. The issue seems to be that the available-endpoints attribute in the power_topology cluster generates a failure:

chip-tool powertopology read available-endpoints 1 1
[1750840555.583] [143091:143091] [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)

We don't actually use that attribute in our project. We tried creating a custom instance of power_topology without it (like you did for the power_measurement cluster), but that didn't solve the problem. 

In the example you published, with matter version 1.5, doesn't seem to include the available-endpoints attribute at all

Do you have any suggestions? Would you be able to try your example on Matter v1.4 to check if you see the same behavior?

Thanks a lot!

jadhavrohit924, 2025-07-01:
When you create your own cluster instance, pass the feature as kNodeTopology and empty optional attributes. This should solve your problem.
Note: Don't forget to create the corresponding features in esp-matter as well, for synchronisation.

```

    gPTDelegate = std::make_unique<PowerTopologyDelegate>();
    gPTInstance =
        std::make_unique<PowerTopologyInstance>(EndpointId(endpoint_id), *gPTDelegate,
                                                BitMask<PowerTopology::Feature, uint32_t>(PowerTopology::Feature::kNodeTopology),
                                                BitMask<PowerTopology::OptionalAttributes, uint32_t>());
    err = gPTInstance->Init();
```

closed 2025-07-25 by crivi30