espressif/esp-matter #1453: execute a factory reset with a button

ravencarcass, 2025-06-11:
Hello, 
I am creating a project with the ESP32C6, and I would like to add an enhancement to a project. 
I would like to add a button that I can press and when pressed: it performs a matter esp factory reset. 
I was wondering if this could be done?
If so, how exactly should I do this. 
I searched for it but didn't really find anything about it.

IDF: ESP-IDF v5.4
edit program: vscode

jadhavrohit924, 2025-06-11:
@ravencarcass We already have the support in all examples. Please check `app_reset_button_register` in `app_main.cpp`

dhrishi, 2025-07-09:
@ravencarcass Please close the issue if this helps

dhrishi, 2025-07-15:
Closing this now. Please re-open in case of any questions/issues

closed 2025-07-15 by dhrishi