espressif/esp-matter #1454: ./chip-tool onoff on-with-timed-off failed

LinKvFM, 2025-06-11:
**Describe the bug**
Use chiptool to send command ./chip-tool onoff on-with-timed-off 1 300 300 1 2 , and an infinite loop occurs after 30 seconds.

**Environment**
- ESP-Matter Commit Id:66d1ef97d820b4ea9ef861e2be85f94149ad338a
- ESP-IDF Commit Id: tag: v5.2.3
- SoC (eg: ESP32 or ESP32-C3): esp32
- Device Logs (Please attach the log [file):  device type is ESP_MATTER_COLOR_TEMPERATURE_LIGHT_DEVICE_TYPE_ID 0x010C
[log.lt-bridge-matter.20250611140139.txt](https://github.com/user-attachments/files/20686830/log.lt-bridge-matter.20250611140139.txt)

[image]

When I cancel the creation of level_control, it seems to become normal.
This is the run log after canceling the creation of level_control: 

[log.lt-bridge-matter.20250611140317.txt](https://github.com/user-attachments/files/20686907/log.lt-bridge-matter.20250611140317.txt)

jadhavrohit924, 2025-06-11:
@LinKvFM This is expected behaviour that it will go in loop till `OnTime` attribute value becomes `ZERO` and level_control cluster has no effect on the command. I have observed the same behaviour with and without level_control cluster. Can you please test it again? If you again see the issue, please add the reproduction steps.

LinKvFM, 2025-06-11:
@jadhavrohit924 If you can see the screenshot I uploaded you can see the value of the OnTime attribute 
`********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x00004001 is 0 **********`

This seems to happen 100% of the time in my case, unless you cancel the creation of level_control. My initial discovery was when testing _TC-OO-2.3_.

jadhavrohit924, 2025-06-11:
@LinKvFM Yes this is supposed to seen every time as this is expected behaviour. But removing level_control should not affect this behaviour is what I'm saying. You should observe same behaviour with/without level_control.

LinKvFM, 2025-07-05:
Update the latest commit to solve