espressif/esp-matter #1460: Matter Bridge Commissioning Fails Due to BLE Termination and PacketBuffer Exhaustion

SYESHASWINI, 2025-06-17:
I'm currently using the bridge_cli example to add non-Matter devices to a Matter bridge. I receive the devices information via UART and create the corresponding endpoints.

When I attempt to commission the bridge, the process fails with the error. Attached the error log file.

[error_log.txt](https://github.com/user-attachments/files/20770554/error_log.txt)

From the logs, it seems that the BLE connection is getting disconnected or terminated. Could someone help me understand what might be causing this and how to resolve it?

**Environment**
- ESP-Matter Commit Id: d144bbb
- ESP-IDF Commit Id: v5.4.1
- SoC (eg: ESP32 or ESP32-C3): ESP32C3
- Device Logs (Please attach the log file): attached

- Host Machine OS: Ubuntu(WSL)
- Host Machine Python version: 3.0

Thanks in advance!

shubhamdp, 2025-06-20:
@SYESHASWINI esp-matter commit you shared is invalid, can you please share the valid commit id that you used.

Also, logs are not complete, can you share the complete logs from the boot up, along with commands that you used to add the bridged device so that complete state of the device can be figured out.

I tried this with esp-matter commit id 7aa49f7, added 4 bridged device using `matter esp bridge add 1 0x100` and was able to commission the device.

Please share more details to debug this further.

SYESHASWINI, 2025-06-20:
Sharing the correct commit ID: ba4b9c6. When I attempt commissioning, it fails with the error: PacketBuffer: pool EMPTY.

I'm attaching the error log for reference. 
[error_log_Commissioning.txt](https://github.com/user-attachments/files/20832233/error_log_Commissioning.txt)

shubhamdp, 2025-06-23:
@SYESHASWINI Thanks for sharing the detailed logs.

Logs shows that device is failing to advertise the mdns instance, can you please enable the debug logging to get more information about why it could be happening.

Try `idf.py menuconfig`, navigate to
```
(Top) → Component config → Log → Log Level → Default log verbosity -> Debug
```

Can you try adding these code lines for monitoring the heap statistics:
```
    while (true) {
        printf("\tDescription\tInternal\tSPIRAM\n");
        printf("Current Free Memory\t%d\t\t%d\n",
               heap_caps_get_free_size(MALLOC_CAP_8BIT) - heap_caps_get_free_size(MALLOC_CAP_SPIRAM),
               heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
        printf("Largest Free Block\t%d\t\t%d\n", heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
               heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
        printf("Min. Ever Free Size\t%d\t\t%d\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
               heap_caps_get_minimum_free_size(MALLOC_CAP_SPIRAM));

        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }
```

If you think the free memory is low, please try few options from here as per your requirement: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/optimizations.html

dhrishi, 2025-08-07:
@SYESHASWINI Please provide the requested information or close the issue if it works.

dhrishi, 2025-09-13:
Closing this now