espressif/esp-matter #1461: V1.4 test fan [TC_FAN_3.3] encountered an error

MaplestoryAlen, 2025-06-17:
My device is a V1.4 matter bridge, and the sub device has a fan under endpiont2. During testing [TC_SAN_3.3], it prompts "chip [DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0202, attributiId: 0x0000_0007err=5c3". I successfully executed the TestPlan step in the chip tool.

log:
 (61682) app_main: Commissioning session started
I (61683) app_main: Commissioning window closed
I (61684) NetAccess: Commissioning Close Timer (re)started for 1 minute
I (61810) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
E (62099) chip[DMG]: Read request on unknown cluster - no data version available
E (62100) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0002err = 5c3
E (62109) chip[DMG]: Read request on unknown cluster - no data version available
E (62117) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0001err = 5c3
E (62129) chip[DMG]: Read request on unknown cluster - no data version available
E (62137) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0000err = 5c3
E (62150) chip[DMG]: Read request on unknown cluster - no data version available
E (62158) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0007err = 5c3
E (62170) chip[DMG]: Read request on unknown cluster - no data version available
E (62178) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0006err = 5c3
I (62405) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (62406) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 4 **********
I (62417) app_driver: endpoint 0,cluster: 48 write : 4  

I (62507) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 4 **********
I (62508) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 5 **********
I (62519) app_driver: endpoint 0,cluster: 48 write : 5  

I (62611) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x0000003E
I (62683) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x0000003E
I (62751) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x0000003E
I (63017) esp_matter_command: Received command 0x00000004 for endpoint 0x0000's cluster 0x0000003E
I (63533) esp_matter_command: Received command 0x0000000B for endpoint 0x0000's cluster 0x0000003E
I (63842) esp_matter_command: Received command 0x00000006 for endpoint 0x0000's cluster 0x0000003E
I (64280) app_main: Fabric is updated
E (66079) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (66841) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 5 **********
I (66842) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (66854) app_driver: endpoint 0,cluster: 48 write : 0  

I (66863) app_main: Fabric is committed
I (66867) app_main: Commissioning complete
I (66869) app_driver:  Network access set paired 

I (66874) esp_matter_core: Commissioning Complete
I (66880) app_main: Commissioning window closed
E (76848) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
E (77631) chip[DMG]: Read request on unknown cluster - no data version available
E (77631) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0202, attributeId: 0x0000_0007err = 5c3
E (96868) chip[SWU]: No suitable OTA Provider candidate found

Matter Test Harness Ver: v2.12+spring2025

TH log:

[UI_Test_Run_2025_06_17_16_38_52.log](https://github.com/user-attachments/files/20771849/UI_Test_Run_2025_06_17_16_38_52.log)

MaplestoryAlen, 2025-06-17:
> Hi [@MaplestoryAlen](https://github.com/MaplestoryAlen) , I'm looking into it till that please share the PICS which you have used during testing this test case.

[Fan Control Cluster Test Plan.zip](https://github.com/user-attachments/files/20772386/Fan.Control.Cluster.Test.Plan.zip)

VaishaliAvhale, 2025-06-17:
Hi @MaplestoryAlen,
You mentioned that you have enabled the Fan on endpoint 2, and I can see that you passed endpoint 2 correctly in the logs. However, the RockSupport attribute is being checked on endpoint 1, which is why the test case failed with "unsupported attribute" and "unsupported cluster" messages.Just for cross check please share Test Harness  project config file here.

[image]

MaplestoryAlen, 2025-06-17:
> You mentioned that you have enabled the Fan on endpoint 2, and I can see that you passed endpoint 2 correctly in the logs. However, the RockSupport attribute is being checked on endpoint 1, which is [...]

Project Config:

{
  "config": {
    "test_parameters": {
      "endpoint": "2"
    },
    "int-arg": "use_pase_only:0",
    "timeout": "10000",
    "network": {
      "wifi": {
        "ssid": "Gerlsair_XizhenElec",
        "password": "AcSmartExp"
      },
      "thread": {
        "rcp_serial_path": "/dev/ttyACM0",
        "rcp_baudrate": 115200,
        "on_mesh_prefix": "fd11:22::/64",
        "network_interface": "eth0",
        "dataset": {
          "channel": "15",
          "panid": "0x1234",
          "extpanid": "1111111122222222",
          "networkkey": "00112233445566778899aabbccddeeff",
          "networkname": "DEMO"
        },
        "otbr_docker_image": null
      }
    },
    "dut_config": {
      "discriminator": "1410",
      "setup_code": "7325600",
      "pairing_mode": "onnetwork",
      "chip_timeout": null,
      "chip_use_paa_certs": true,
      "trace_log": true,
      "enhanced_setup_flow": null
    }
  },
  "pics": {
    "clusters": {
      "Fan Control Cluster Test Plan": {
        "name": "Fan Control Cluster Test Plan",
        "items": {
          "FAN.S": {
            "number": "FAN.S",
            "enabled": true
          },
          "FAN.C": {
            "number": "FAN.C",
            "enabled": false
          },
          "FAN.S.A0000": {
            "number": "FAN.S.A0000",
            "enabled": true
          },
          "FAN.S.A0001": {
            "number": "FAN.S.A0001",
            "enabled": true
          },
          "FAN.S.A0002": {
            "number": "FAN.S.A0002",
            "enabled": true
          },
          "FAN.S.A0003": {
            "number": "FAN.S.A0003",
            "enabled": true
          },
          "FAN.S.A0004": {
            "number": "FAN.S.A0004",
            "enabled": true
          },
          "FAN.S.A0005": {
            "number": "FAN.S.A0005",
            "enabled": true
          },
          "FAN.S.A0006": {
            "number": "FAN.S.A0006",
            "enabled": true
          },
          "FAN.S.A0007": {
            "number": "FAN.S.A0007",
            "enabled": true
          },
          "FAN.S.A0008": {
            "number": "FAN.S.A0008",
            "enabled": true
          },
          "FAN.S.A0009": {
            "number": "FAN.S.A0009",
            "enabled": false
          },
          "FAN.S.A000a": {
            "number": "FAN.S.A000a",
            "enabled": false
          },
          "FAN.S.A000b": {
            "number": "FAN.S.A000b",
            "enabled": false
          },
          "FAN.S.C00.Rsp": {
            "number": "FAN.S.C00.Rsp",
            "enabled": false
          },
          "FAN.S.F00": {
            "number": "FAN.S.F00",
            "enabled": true
          },
          "FAN.S.F01": {
            "number": "FAN.S.F01",
            "enabled": true
          },
          "FAN.S.F02": {
            "number": "FAN.S.F02",
            "enabled": true
          },
          "FAN.S.F03": {
            "number": "FAN.S.F03",
            "enabled": false
          },
          "FAN.S.F04": {
            "number": "FAN.S.F04",
            "enabled": false
          },
          "FAN.S.F05": {
            "number": "FAN.S.F05",
            "enabled": false
          }
        }
      }
    }
  }
}

VaishaliAvhale, 2025-06-17:
The project configuration is correct; the issue seems to be with the Python script.
Could you please confirm whether you're on this commit: 78cc902705428bbb56f7fc37ac72f33712787c11?

MaplestoryAlen, 2025-06-18:
> The project configuration is correct.the issue lies in the Python script, which was fixed in November 2024. You can refer to the fix here: 🔗 [project-chip/connectedhomeip@926ab43](https://github.com/project-chip/connectedhomeip/commit/926ab430801b4158a8d75b1daf1b1adccf230639)
> 
> Please raise an issue here to request backporting of this change to the v1.4 Certification branch: 🔗 https://github.com/project-chip/certification-tool/issues
> 
> Note: If you're verifying your application for v1.4, you're currently using the incorrect certification branch. Please use the correct branch as mentioned in the official documentation: 🔗 https://github.com/project-chip/certification-tool/blob/main/docs/Matter_TH_User_Guide/Matter_TH_User_Guide.adoc
> 
> The issue also exists in the current v1.4 Test Harness branch, which is why I recommended raising the issue.

Thank you very much for your answer