espressif/esp-matter #1462: TC-IDM-10.2 Battery storage endpoint test failed

bilalahmaddev, 2025-06-17:
**Describe the bug**
[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:11 (0x0b)
    problem: Attribute 0x0b is included, but is disallowed by conformance. Conformance: [BAT], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:12 (0x0c)
    problem: Attribute 0x0c is included, but is disallowed by conformance. Conformance: [BAT], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:13 (0x0d)
    problem: Attribute 0x0d is included, but is disallowed by conformance. Conformance: [BAT], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:18 (0x12)
    problem: Attribute 0x12 is included, but is disallowed by conformance. Conformance: [BAT], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:24 (0x18)
    problem: Attribute 0x18 is included, but is disallowed by conformance. Conformance: [REPLC | RECHG], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:27 (0x1b)
    problem: Attribute 0x1b is included, but is disallowed by conformance. Conformance: [RECHG], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:29 (0x1d)
    problem: Attribute 0x1d is included, but is disallowed by conformance. Conformance: [RECHG], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:30 (0x1e)
    problem: Attribute 0x1e is included, but is disallowed by conformance. Conformance: [RECHG], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:26 (0x1a)
    problem: Attribute 0x1a is included, but is disallowed by conformance. Conformance: RECHG, implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:14 (0x0e)
    problem: Attribute 0x0e is included, but is disallowed by conformance. Conformance: BAT, implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:15 (0x0f)
    problem: Attribute 0x0f is included, but is disallowed by conformance. Conformance: BAT, implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:16 (0x10)
    problem: Attribute 0x10 is included, but is disallowed by conformance. Conformance: BAT, implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:17 (0x11)
    problem: Attribute 0x11 is included, but is disallowed by conformance. Conformance: [BAT], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:20 (0x14)
    problem: Attribute 0x14 is included, but is disallowed by conformance. Conformance: [REPLC], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:21 (0x15)
    problem: Attribute 0x15 is included, but is disallowed by conformance. Conformance: [REPLC], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:22 (0x16)
    problem: Attribute 0x16 is included, but is disallowed by conformance. Conformance: [REPLC], implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:25 (0x19)
    problem: Attribute 0x19 is included, but is disallowed by conformance. Conformance: REPLC, implemented features: WIRED
    spec_location: 

[MatterTest] 06-17 09:03:28.253 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  47 (0x2f) PowerSource
      Attribute:28 (0x1c)
    problem: Attribute 0x1c is included, but is disallowed by conformance. Conformance: RECHG, implemented features: WIRED
    spec_location: 

**Environment**
- ESP-Matter Commit Id: matter release 1.4
- ESP-IDF Commit Id: 5.41
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3

VaishaliAvhale, 2025-06-17:
Please share the complete TH logs and the PICS details of the applications present on Endpoint 19.

bilalahmaddev, 2025-06-17:
@VaishaliAvhale I found the issue and added a fix for it and pull request is created (https://github.com/espressif/esp-matter/pull/1464) .  Test case is passing after adding this fix. 

This issue is caused by a bug in the Power Source cluster's feature settings . Previously, the feature checks for Battery and Wired were mistakenly swapped, leading to incorrect validation of the feature for wire and battery (https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_feature.cpp) . As a result, the Battery feature was not correctly enabled in the Power Source cluster, and without this feature for power source cluster, battery-related attributes such as voltage and percentage are not allowed.