espressif/esp-matter #1465: Matter Window Covering Device Reporting Random Values

SYESHASWINI, 2025-06-18:
Hi everyone,

I’ve created a window_covering device on my Matter bridge using the following code:

> window_covering_device::config_t` window_config;
> 
>  endpoint_t *endpoint_4 = window_covering_device::create(node, &window_config, ENDPOINT_FLAG_NONE, nullptr);
> 
> /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
> if (!node || !endpoint_4) {
>     ESP_LOGE(TAG, "Matter node creation failed");
> }
> 
> /* Add additional features to the node */
> cluster_t *cluster = cluster::get(endpoint_4, chip::app::Clusters::WindowCovering::Id);
> 
> cluster::window_covering::feature::lift::config_t lift;
> cluster::window_covering::feature::tilt::config_t tilt;
> cluster::window_covering::feature::position_aware_lift::config_t position_aware_lift;
> cluster::window_covering::feature::position_aware_tilt::config_t position_aware_tilt;
> cluster::window_covering::feature::absolute_position::config_t absolute_position;
> 
> cluster::window_covering::feature::lift::add(cluster, &lift);
> cluster::window_covering::feature::tilt::add(cluster, &tilt);
> cluster::window_covering::feature::position_aware_lift::add(cluster, &position_aware_lift);
> cluster::window_covering::feature::position_aware_tilt::add(cluster, &position_aware_tilt);
> cluster::window_covering::feature::absolute_position::add(cluster, &absolute_position);

The device gets added successfully and shows up in the network. However, when I operate it, the values seem random and inconsistent — not following the expected `0–100/0-255` range. The behavior is unpredictable, and the device is often marked as not responsive.

Has anyone else experienced similar behavior? Am I missing something in the configuration or initialization steps?

Any help or pointers would be appreciated.

Thanks!

SYESHASWINI, 2025-06-19:
I'm trying to open and close, but the values are changing randomly during operation. When I operate it, I get:
**Cluster = 0x102, Attribute ID = 0xB**

I've also added a fan and a dimmable light, and both are working well, so I suspect it is not a network issue.

jadhavrohit924, 2025-06-20:
I have tested it with Google home and it is working fine. Which controller you are using? Can you please share the complete logs in a file?

jadhavrohit924, 2025-07-25:
@SYESHASWINI Do you have any updates? Please close the issue if your it is resolved.

dhrishi, 2025-08-20:
Closing this now. Please re-open by providing the anticipated information

closed 2025-08-20 by dhrishi