espressif/esp-matter #1466: TC-IDM-10.5 Extra cluster found on endpoint with device types (Solar Power and EVSE)

bilalahmaddev, 2025-06-18:
TC-IDM-10.5 Extra cluster found on endpoint with device types (Solar Power and EVSE) But as per specification both can have device energy measurement clusters. 

[MatterTest] 06-18 09:17:38.593 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.5
    location: 
       Endpoint: 6,
       Cluster:  152 (0x98) DeviceEnergyManagement
    problem: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=23, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=17, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=1296, revision=1)]
    spec_location: 

[MatterTest] 06-18 09:17:38.593 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.5
    location: 
       Endpoint: 15,
       Cluster:  152 (0x98) DeviceEnergyManagement
    problem: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=1292, revision=2)]
    spec_location: 

[TC_IDM_10_5_2025_06_18_14_16_37.log](https://github.com/user-attachments/files/20792885/TC_IDM_10_5_2025_06_18_14_16_37.log)

Matter release 1.4
MHT: Version: v2.11+fall2024 Sha: 8f86ccd
esp32-s3

VaishaliAvhale, 2025-06-18:
Hi @bilalahmaddev,
Could you please share the PICS for endpoint 6 and endpoint 15? I need them for issue analysis.

bilalahmaddev, 2025-06-18:
Ok I will share. Can you confirm if this is related to this issue:
https://github.com/project-chip/connectedhomeip/issues/36827

jadhavrohit924, 2025-06-19:
DeviceEnergyManagement is mandatory for EVSE and optional for Solar Power device types in the spec, did you change anything? Is you have added the cluster on Solar Power you should update the same in the PICS files as well.

VaishaliAvhale, 2025-06-19:
Hi @bilalahmaddev, In the first issue, I see Device Type ID 1296, which corresponds to an Electrical Sensor. For this device type, the Device Energy Management cluster is neither mandatory nor optional, so the issue is valid. Please correct the firmware.

In the second issue, the device is EVSE, for which Device Energy Management is a mandatory cluster. So this is an issue on the Python script side. Please raise it here: https://github.com/project-chip/connectedhomeip/issues

VaishaliAvhale, 2025-06-19:
Yes looking into endpoint 6 issue.

>> Also can you please share correct pull request link?

My bad, updated it now.

bilalahmaddev, 2025-06-19:
@VaishaliAvhale This pull request (https://github.com/project-chip/connectedhomeip/pull/36201) does not have any python script update for TC-IDM-10.5

[image]

VaishaliAvhale, 2025-06-19:
@bilalahmaddev, I just went through one Slack discussion related to endpoint 6 issue in that Cecille Freeman said "Extra clusters are technically permitted at cert, it's just a warning, but if you check the spec, it's a decently strong warning because it's totally possible that clients will just ignore clusters that are not marked by a device type. "

Here is her second comment

[image]

VaishaliAvhale, 2025-06-19:
Hi @bilalahmaddev,
Could you please raise the second issue here: https://github.com/project-chip/connectedhomeip/issues?
Since the Device Energy Management cluster is mandatory, it should not be marked as an extra cluster.

bilalahmaddev, 2025-06-19:
@VaishaliAvhale issue created on connectedhomeip: https://github.com/project-chip/connectedhomeip/issues/39613