espressif/esp-matter #1467: TC-G-2.2: max-groups-per-fabric attribute does not account for dynamic end points

mikesolsen, 2025-06-19:
**Describe the bug**
`esp_matter_chip_init_task()` initializes `max-groups-per-fabric` once at boot. In the case of a bridge, new devices may be discovered and added after calling `esp_matter::start()`.  If bridged end points with the group cluster are added after calling `esp_matter::start()`, the limit is not increased.  

**Environment**
- ESP-Matter Commit Id: f353b9e3d400ad80b254c3594e57154f8fa32ee9
- ESP-IDF Commit Id: v5.4.1 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):

- Host Machine OS: n/a to this issue
- Host Machine Python version: n/a to this issue

- Commissioner app and versions if present: n/a
- Home hub app and versions if present: n/a
- Commissioner's logs if present: n/a

**Any additional details**
In my case, I had zero bridged devices known to the bridge after a factory reset. After adding multiple devices, the max-groups-per-fabric attribute remained at 0. This causes TC-G-2.2 to fail.

Looks like the current approach arose from this issue: https://github.com/espressif/esp-matter/issues/724, 
which was implemented in this commit: https://github.com/espressif/esp-matter/commit/974997de95dba6478b56cfba40f71fd59e0ba06d
Also relevant: https://github.com/espressif/esp-matter/issues/1236

shubhamdp, 2025-06-23:
@mikesolsen Thanks for reporting, checking if we can dynamically change the group data provider so that it can adapt to the newly added bridged device.

shubhamdp, 2025-06-24:
@mikesolsen could you try with the attached patch. I have re-run the TC-G-2.2 and works as expected with bridged endpoints.

I used the [bridge-cli](https://github.com/espressif/esp-matter/tree/main/examples/bridge_apps/bridge_cli) app for adding/removing dynamic endpoints...
```
matter esp bridge add 1 0x100
matter esp bridge add 1 0x100
matter esp bridge add 1 0x100
matter esp bridge add 1 0x100
```

```
curl https://github.com/espressif/esp-matter/commit/fcedfa40e9fde2bcf9e5f674e11199e852f51021.patch | git am
```

mikesolsen, 2025-06-24:
@shubhamdp, this is working well for me.  I was able to complete TC-G-2.2 successfully after a factory reset of the bridge. Additionally, I added and removed bridged endpoints at various steps in TC-G-2.2 without issue.

Thanks for looking into this quickly.
Mike

mikesolsen, 2025-06-26:
@shubhamdp Unfortunately we moved a bit fast.  It appears this patch is causing a crash upon executing a group command.

By following the steps at this link (slightly adapted to have 2 bridged end points in the group), it crashes. I've attached the backtrace.  If I revert the patch, then the group command succeeds.

https://project-chip.github.io/connectedhomeip-doc/examples/chip-tool/README.html#configuring-the-server-side-for-group-commands

[crash backtrace.txt](https://github.com/user-attachments/files/20930374/crash.backtrace.txt)

shubhamdp, 2025-06-27:
@mikesolsen Could you re-open this issue. Can you please share the endpoint composition of your device (Is this ep0: root, ep1: aggregator, ep2: bridged-device, ep3: bridged-device ?).

What would help is if you have logs from TH/chip-tool console, please share it so that I can use those commands and can link the similarities that could crash if any.

mikesolsen, 2025-06-27:
Hi @shubhamdp, I don't have the permssions to reopen the github issue.

This is correct: ep0: root, ep1: aggregator, ep2 to ep34 inclusive: bridged-device. I am only using the first 2 bridged devices to reproduce this. Bridged devices are being added after `esp_matter:start()` would have been called.

I've attached logs from chip tool and the exact commands I use to trigger the issue.
[chip-tool logs.txt](https://github.com/user-attachments/files/20953666/chip-tool.logs.txt)
[chip-tool command summary.txt](https://github.com/user-attachments/files/20953665/chip-tool.command.summary.txt)

While reproducing these logs I found one more important detail:
- If I run these commands against the unpatched firmware, the group toggle command works immediately and the bridge end points toggle their state as expected.
- If I run these commands against the patched firmware, the group toggle command appears to be sent by chip-tool, however nothing happens within the bridge (no activity, no logs). If I power cycle the bridge, wait for boot to complete, then issue the group toggle command, the crash occurs.

dhrishi, 2025-06-30:
Re-opening the issue. Thanks @mikesolsen

mikesolsen, 2025-07-25:
@shubhamdp @dhrishi 
Is there any movement on this ticket? I cannot realistically go to certification with this unresolved.

shubhamdp, 2025-07-30:
@mikesolsen I followed the link but commands mentioned there seems incomplete so stick with what you have provided in summary file. Tried to re-produce the crash but was unable to do so.

esp-idf: v5.4.1
esp-matter: [release/v1.4](https://github.com/espressif/esp-matter/tree/40ca7ea9c3ae485ea678a3a40676a2915f8bbe0d) (+ c91d9d5aa7ebd9fc4d0636bc5c6b3863110a2e4e as patch)

I had a weird observation though, where on first boot, ESP32 do not receive any group command (this is the same boot cycle where groups get added).  But, on reboot, device did not crash but I was able to send the group commands to the device.

--

About the crash, it is dereferencing the nullptr. This could be happening when resizing the group data provider.

--

To unblock you, with we still debugging the problem, you can remove these lines and set [CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC=3](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L161-L165) to the desirable value.

https://github.com/espressif/esp-matter/blob/40ca7ea9c3ae485ea678a3a40676a2915f8bbe0d/components/esp_matter/esp_matter_core.cpp#L744-L756

mikesolsen, 2025-07-30:
Thanks for taking a look @shubhamdp . I have hard coded a larger number for max_groups_per_fabric to unblock myself. While this allows TC-G-2.2 to succeed, it results in TC-RR-1.1 failing, as described in this ticket: https://github.com/espressif/esp-matter/issues/1507.

So far, the only resolution provided is to assume the ATL will not request to test with a large number of bridged end points.