espressif/esp-matter #1468: chipDie on restart after invoking change-to-mode from chip-tool

tomasmcguinness, 2025-06-21:
**Describe the bug**
I am attempting to implement a Dishwasher device type. 

```
dish_washer::config_t dish_washer_config;
dish_washer_config.operational_state.delegate = &operational_state_delegate; // Set to nullptr if not using a delegate

endpoint_t *endpoint = dish_washer::create(node, &dish_washer_config, ENDPOINT_FLAG_NONE, NULL);
```

I have added a Dishwasher mode cluster and some attributes and commands

```
static DishwasherModeDelegate dish_washer_mode_delegate;

esp_matter::cluster::dish_washer_mode::config_t dish_washer_mode_config;
dish_washer_mode_config.delegate = &dish_washer_mode_delegate;
dish_washer_mode_config.current_mode = DishwasherMode::ModeNormal; // Set the initial mode

esp_matter::cluster_t *dish_washer_mode_cluster = esp_matter::cluster::dish_washer_mode::create(endpoint, &dish_washer_mode_config, CLUSTER_FLAG_SERVER);
ABORT_APP_ON_FAILURE(dish_washer_mode_cluster != nullptr, ESP_LOGE(TAG, "Failed to create dishwashermode cluster"));

esp_matter::cluster::mode_base::attribute::create_supported_modes(dish_washer_mode_cluster, NULL, 0, 0);

esp_matter::cluster::mode_base::command::create_change_to_mode(dish_washer_mode_cluster);
```

When I pair the device with the chip-tool, I can call the `change-to-mode` command and receive a successful response

```
chip-tool dishwashermode change-to-mode 0x02 0x06 0x01

yields

[1750496240.477] [182588:182619] [DMG] Received Command Response Data, Endpoint=1 Cluster=0x0000_0059 Command=0x0000_0001
[1750496240.477] [182588:182619] [TOO] Endpoint: 1 Cluster: 0x0000_0059 Command 0x0000_0001
[1750496240.477] [182588:182619] [TOO]   ChangeToModeResponse: {
[1750496240.477] [182588:182619] [TOO]     status: 0
[1750496240.477] [182588:182619] [TOO]    }
```

However, as soon as I reset the board (using the button on the dev kit or unplugging the USB cable), it immediately goes into a crash loop.

```
I (2421) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x0000FFFC is 0 **********
I (2431) esp_matter_core: Dynamic endpoint 1 added
E (2441) chip[IM]: Duplicate command handler registration failed
I (2441) app_driver: DishwasherModeDelegate::GetModeValueByIndex(0)
I (2451) app_driver: DishwasherModeDelegate::GetModeValueByIndex(0)
I (2461) app_driver: DishwasherModeDelegate::GetModeValueByIndex(1)
I (2471) app_driver: DishwasherModeDelegate::GetModeValueByIndex(2)
E (2471) chip[DL]: Chip stack locking error at '/home/tomasmcguinness/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/reporting.cpp:44'. Code is unsafe/racy
E (2491) chip[-]: chipDie chipDie chipDie
```

```
abort() was called at PC 0x420540ef on core 0
--- 0x420540ef: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/tomasmcguinness/development/matter-esp32-toy-dishwasher/build/esp-idf/chip/../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:484
---  (inlined by) chipDie at /home/tomasmcguinness/development/matter-esp32-toy-dishwasher/build/esp-idf/chip/../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:494
---  (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/tomasmcguinness/development/matter-esp32-toy-dishwasher/build/esp-idf/chip/../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36
```

The back trace is 

```
panic_abort (details=details@entry=0x4083a16c "abort() was called at PC 0x420540ef on core 0") at /home/tomasmcguinness/esp/esp-idf/components/esp_system/panic.c:468
468         asm("unimp");   // should be an invalid operation on RISC-V targets
#0  panic_abort (details=details@entry=0x4083a16c "abort() was called at PC 0x420540ef on core 0") at /home/tomasmcguinness/esp/esp-idf/components/esp_system/panic.c:468
#1  0x4080d298 in esp_system_abort (details=details@entry=0x4083a16c "abort() was called at PC 0x420540ef on core 0") at /home/tomasmcguinness/esp/esp-idf/components/esp_system/port/esp_system_chip.c:87
#2  0x408142b4 in abort () at /home/tomasmcguinness/esp/esp-idf/components/newlib/abort.c:38
#3  0x420540f2 in chipAbort () at ../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:484
#4  chipDie () at ../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:494
#5  chip::Platform::Internal::AssertChipStackLockedByCurrentThread (file=file@entry=0x42158094 "", line=line@entry=44) at ../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36
#6  0x4201e3da in MatterReportingAttributeChangeCallback (aPath=...) at /home/tomasmcguinness/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/reporting.cpp:44
#7  0x420237e4 in chip::app::Clusters::ModeBase::Instance::UpdateCurrentMode (this=this@entry=0x40856434, aNewMode=<optimized out>) at /home/tomasmcguinness/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-base-server/mode-base-server.cpp:179
#8  0x42023b6a in chip::app::Clusters::ModeBase::Instance::LoadPersistentAttributes (this=this@entry=0x40856434) at /home/tomasmcguinness/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-base-server/mode-base-server.cpp:432
#9  0x42023c48 in chip::app::Clusters::ModeBase::Instance::Init (this=0x40856434) at /home/tomasmcguinness/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-base-server/mode-base-server.cpp:79
#10 0x42012e44 in esp_matter::cluster::delegate_init_callback_common () at /home/tomasmcguinness/esp/esp-matter/components/esp_matter/esp_matter_cluster.cpp:80
#11 0x42014b24 in esp_matter::chip_init (callback=0x4200d222 <app_event_cb(chip::DeviceLayer::ChipDeviceEvent const*, intptr_t)>, callback_arg=0) at /home/tomasmcguinness/esp/esp-matter/components/esp_matter/esp_matter_core.cpp:892
#12 esp_matter::start (callback=0x4200d222 <app_event_cb(chip::DeviceLayer::ChipDeviceEvent const*, intptr_t)>, callback_arg=0) at /home/tomasmcguinness/esp/esp-matter/components/esp_matter/esp_matter_core.cpp:916
#13 0x4200d7b8 in app_main () at /home/tomasmcguinness/development/matter-esp32-toy-dishwasher/main/app_main.cpp:164
#14 0x421495a8 in main_task (args=<error reading variable: value has been optimized out>) at /home/tomasmcguinness/esp/esp-idf/components/freertos/app_startup.c:208
#15 0x4080e1cc in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /home/tomasmcguinness/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:255
```

I understand this issue is caused by the fact the Matter stack is not locked, but I'm unable to determine where I should be locked it.

One other question

In the `HandleChangeToMode` of the DishwasherModeDelegate, the mode parameter is a uint8_t, yet some of the ModeTag values are based off uint16_t e.g. ModeTag::kNormal. Is the `mode` parameter supposed to be the index?

**Environment**
- ESP-Matter Commit Id: ba4b9c6bcde2469ec081759974a80bf670a7bc54
- ESP-IDF Commit Id: 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):

- Host Machine OS: Windows WLS
- Host Machine Python version: 3.10.12

- Commissioner app and versions if present: iOS and chip-tool
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

tomasmcguinness, 2025-06-22:
It also looks like two instances of the ModeBase Cluster for endpoint 1 are being created.

[image]

I don't know if that makes a difference, but it's certainly unexpected.

I'm following the examples from the chip examples. I think I should not need to define my own `emberAfDishwasherModeClusterInitCallback` as it looks like either esp-matter or chipSDK is doing it for me.

jadhavrohit924, 2025-06-23:
Thanks for the report and investigation @tomasmcguinness. You should use either esp-matters `DishWasherModeDelegateInitCB` or `emberAfDishwasherModeClusterInitCallback`. There is locking issue with `DishWasherModeDelegateInitCB` we will fix it soon.

Thank You!

closed 2025-06-30 by espressif-bot

dhrishi, 2025-06-30:
@tomasmcguinness Thank you for reporting the issue. The fix is now available in the above mentioned commit. Please check if it helps.