espressif/esp-matter #1471: TC-EEVSE-2.2 UserMaximumChargeCurrent attribute write failed,

bilalahmaddev, 2025-06-24:
**Describe the bug**

 TC-EEVSE-2.2 UserMaximumChargeCurrent write failed and getting "status" : "136 == kUnsupportedWrite" in test

<<< [E:2976i S:2400 M:39642590] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [4B50] [UDP:192.168.18.136:5540] --- Type 0001:06 (IM:WriteRequest) (B:62)
[MatterTest] 06-24 16:28:44.051 INFO ??1 [E:2976i S:2400 M:39642590] (S) Msg Retransmission to 1:0000000012344321 in 397ms [State:Active II:500 AI:300 AT:4000]
[MatterTest] 06-24 16:28:44.138 INFO {
[MatterTest] 06-24 16:28:44.139 INFO 	"event" : "MessageReceived",
[MatterTest] 06-24 16:28:44.139 INFO 	"messageType" : "Secure",
[MatterTest] 06-24 16:28:44.139 INFO 	"packetHeader" : 
[MatterTest] 06-24 16:28:44.139 INFO 	{
[MatterTest] 06-24 16:28:44.139 INFO 		"flags" : 0,
[MatterTest] 06-24 16:28:44.139 INFO 		"msgCounter" : 157738660,
[MatterTest] 06-24 16:28:44.139 INFO 		"securityFlags" : 0,
[MatterTest] 06-24 16:28:44.139 INFO 		"sessionId" : 2400
[MatterTest] 06-24 16:28:44.140 INFO 	},
[MatterTest] 06-24 16:28:44.140 INFO 	"payload" : 
[MatterTest] 06-24 16:28:44.140 INFO 	{
[MatterTest] 06-24 16:28:44.140 INFO 		"decoded" : 
[MatterTest] 06-24 16:28:44.140 INFO 		{
[MatterTest] 06-24 16:28:44.140 INFO 			"write_response" : 
[MatterTest] 06-24 16:28:44.140 INFO 			{
[MatterTest] 06-24 16:28:44.140 INFO 				"interaction_model_revison" : "12",
[MatterTest] 06-24 16:28:44.140 INFO 				"write_responses" : 
[MatterTest] 06-24 16:28:44.140 INFO 				{
[MatterTest] 06-24 16:28:44.140 INFO 					"Anonymous<0>" : 
[MatterTest] 06-24 16:28:44.141 INFO 					{
[MatterTest] 06-24 16:28:44.141 INFO 						"path" : 
[MatterTest] 06-24 16:28:44.141 INFO 						{
[MatterTest] 06-24 16:28:44.141 INFO 							"attribute_id" : "9 == 'userMaximumChargeCurrent'",
[MatterTest] 06-24 16:28:44.141 INFO 							"cluster_id" : "153 == 'EnergyEvse'",
[MatterTest] 06-24 16:28:44.141 INFO 							"endpoint_id" : "15"
[MatterTest] 06-24 16:28:44.141 INFO 						},
[MatterTest] 06-24 16:28:44.141 INFO 						"status" : 
[MatterTest] 06-24 16:28:44.141 INFO 						{
[MatterTest] 06-24 16:28:44.141 INFO 							"status" : "136 == kUnsupportedWrite"
[MatterTest] 06-24 16:28:44.141 INFO 						}
[MatterTest] 06-24 16:28:44.142 INFO 					}
[MatterTest] 06-24 16:28:44.142 INFO 				}
[MatterTest] 06-24 16:28:44.142 INFO 			}
[MatterTest] 06-24 16:28:44.142 INFO 		},
[MatterTest] 06-24 16:28:44.142 INFO 		"hex" : "15360015370024020F24039924040918350124008818181824FF0C18",
[MatterTest] 06-24 16:28:44.142 INFO 		"size" : 28
[MatterTest] 06-24 16:28:44.142 INFO 	},
[MatterTest] 06-24 16:28:44.142 INFO 	"payloadHeader" : 
[MatterTest] 06-24 16:28:44.142 INFO 	{
[MatterTest] 06-24 16:28:44.142 INFO 		"ackMessageCounter" : 39642590,
[MatterTest] 06-24 16:28:44.142 INFO 		"exchangeFlags" : 6,
[MatterTest] 06-24 16:28:44.143 INFO 		"exchangeId" : 2976,
[MatterTest] 06-24 16:28:44.143 INFO 		"initiator" : false,
[MatterTest] 06-24 16:28:44.143 INFO 		"messageType" : 7,
[MatterTest] 06-24 16:28:44.143 INFO 		"needsAck" : true,
[MatterTest] 06-24 16:28:44.143 INFO 		"protocolId" : 1
[MatterTest] 06-24 16:28:44.143 INFO 	}
[MatterTest] 06-24 16:28:44.143 INFO }
[MatterTest] 06-24 16:28:44.143 INFO >>> [E:2976i S:2400 M:157738660 (Ack:39642590)] (S) Msg RX from 1:0000000012344321 [4B50] to 000000000001B669 --- Type 0001:07 (IM:WriteResponse) (B:62)
[MatterTest] 06-24 16:28:44.146 INFO {
[MatterTest] 06-24 16:28:44.146 ERROR Exception occurred in test_TC_EEVSE_2_2.
Traceback (most recent call last):

**Environment**
- ESP-Matter Commit Id:
1.4
- ESP-IDF Commit Id:
- 5.41
- SoC (eg: ESP32 or ESP32-C3):
ESP32-S3
- Device Logs (Please attach the log file):

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

Jerry-ESP, 2025-06-25:
For this attribute, you can change here as below: 
https://github.com/espressif/esp-matter/blob/1c7271b8acca01b2d71c77948c84ce431ebe9797/components/esp_matter/esp_matter_attribute.cpp#L4847 

attribute_t *create_user_maximum_charge_current(cluster_t *cluster, int64_t value)
{
    return esp_matter::attribute::create(cluster, EnergyEvse::Attributes::UserMaximumChargeCurrent::Id, ATTRIBUTE_FLAG_NONVOLATILE | ATTRIBUTE_FLAG_WRITABLE, esp_matter_int64(value));
}

we will fix this type of issues soon

bilalahmaddev, 2025-06-25:
@Jerry-ESP I tried this and still getting same issue. This is a certification blocker for us.

bilalahmaddev, 2025-06-25:
Previous results was from TC-EEVSE-2.2 test and this is from chip-tool.

Jerry-ESP, 2025-06-26:
Please try to write this attribute through chip-tool with the previous firmware(without ATTRIBUTE_FLAG_WRITABLE). 

And can you share your delegate implementation?

Jerry-ESP, 2025-07-02:
> On my side, after add the ATTRIBUTE_FLAG_WRITABLE, I can write this attribute successfully through chip-tool.
> 
> Please try to write this attribute through chip-tool with the previous firmware(without ATTRIBUTE_FLAG_WRITABLE).
> 
> And can you share your delegate implementation?

@bilalahmaddev  I have tried on my side before. Maybe you missed this message.

Jerry-ESP, 2025-07-02:
> [@Jerry-ESP](https://github.com/Jerry-ESP) maybe I am using matter 1.4.1 as a component in esp-idf ?

Did you ever test with the ATTRIBUTE_FLAG_WRITABLE flag?

Jerry-ESP, 2025-07-02:
We will update the esp-matter component version after this fix merged.

bilalahmaddev, 2025-07-07:
@Jerry-ESP I found the issue, only setting ATTRIBUTE_FLAG_WRITABLE does not work. I am setting delegate through this

jadhavrohit924, 2025-07-07:
@bilalahmaddev `EnergyEvseDelegateInitCB` is an example implementation, if you want to override its default definition directly use custom instance of EnergyEvse as discussed here for [ElectricalPowerMeasurement](https://github.com/espressif/esp-matter/issues/1451)

bilalahmaddev, 2025-07-07:
@jadhavrohit924 The EnergyEvseDelegateInitCB is not merely an example implementation; it is part of the official ESP-Matter library. According to the delegate implementation guide provided here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/app_guide.html, it is recommended to implement and set the delegate using this function.

If it is intended to serve only as an example, this should be clearly stated in the documentation to avoid confusion and unintended use in production code. But esp-matter is not just for example case I think.

jadhavrohit924, 2025-07-15:
@bilalahmaddev We will fix this issue soon.

closed 2025-08-20 by espressif-bot